osqzss/gps-sdr-sim #117: variable meaning

kasmf, 2018-01-17:
what is the variable iumd and numd meaning in the gps-sdr-sim code ... line or have in the generate baseband signals

osqzss, 2018-01-17:
The number of user data motion records.

kasmf, 2018-01-17:
thank u

kasmf, 2018-01-17:
what is sinTable512[] ? how to set this table and what is its function?

osqzss, 2018-01-17:
Try google with "lookup table".

kasmf, 2018-01-17:
what is g.sec in tk = g.sec - eph.toe.sec? by me g.sec is the gps receiption time on the receiver,... i am right or not?

kasmf, 2018-01-17:
it is not clear for me please ekold = ek + 1.0? why ekold is greater than ek by 1.0? and other issue is OneMinusecosE = 0 .... this mean when not get to while loop OneMinusecosE is always zero ... so all the variables divide by OneMinusecosE is zero ... is it right or not?

Honky103, 2018-01-17:
If i am not mistaken g.sec is the time of transmission of the gps signal by the satellites. 

Strongly recommend you read through the ICD document to understand the operations in your latest question

osqzss, 2018-01-17:
Take a look at section 20.3.3.4.3 and Table 20-IV of ICD-200C. If you want to learn more about GNSS, it is advisable to get a decent reference book. I personally recommend Kaplan's "Understanding GPS/GNSS" or Misra's "Global Positioning System".

kasmf, 2018-01-17:
thank u both honky and osqzss .... what is the difference b/n g.sec and eph.toe.sec in above honky?

Honky103, 2018-01-17:
g.sec is the time of transmission. eph.toe.sec is the time of ephemeris

kasmf, 2018-01-17:
what is time of ephemeris? it mean the time it take to generate ephemeris data?

Honky103, 2018-01-17:
I don't think this is the right platform to discuss GPS theory. Please read in depth what osqzss has recommended. You will find most of your answers there.

bikilay, 2018-01-18:
iumd = initial user motion data.... the data of the user at initial time  and umd = user motion data

kasmf, 2018-01-18:
what is  the difference between time of ephemeris 'toe' and time of clock 'toc' in the gps-sdr-sim code?

osqzss, 2018-01-18:
Read section 20.3.3.3.3.1 for TOC and Table 20-IV for TOE.

kasmf, 2018-01-23:
what is number of sets ephemerides in the readRinexNavAll commented line?

osqzss, 2018-01-23:
A GPS ephemeris is valid for a couple of hours. If reading a 24-hour Rinex navigation file, you will get several sets of ephemerides for each satellite.

kasmf, 2018-01-23:
thank u osqzss. what is EPHEM_ARRAY_SIZE?

osqzss, 2018-01-23:
The maximum number of ephemerides for each satellite in a daily GPS broadcast ephemeris file. You have twelve sets of valid 2-hour ephemerides for a day.

kasmf, 2018-01-23:
why initialize g0.week = -1 in the readRinexNavAll ? is it possible g0.week = -1? and when it can be ... it is not clear for me b/c time is can not be negative.

osqzss, 2018-01-23:
It indicates an "invalid" ephemeris. After the function read the navigation file, every valid set of ephemeris will have a positive week number.

kasmf, 2018-01-23:
thank u very much ... osqzss

kasmf, 2018-01-24:
why this and others idot = (long)(eph.idot/POW2_M43/PI) divide by PI? what is the unit when divide by PI?  (hint rad or degree)

Honky103, 2018-01-24:
Have you read the sections suggested by osqzss? They will answer your question on why it needs to be divisible by PI.

Honky103, 2018-01-24:
My apologises, "divided by PI"

osqzss, 2018-01-24:
As it shown in ICD, the unit of angular measure is semi-circle (normalized by a half-cycle).

closed 2018-02-06 by osqzss