osqzss/gps-sdr-sim #118: ek and ekold

robellebor, 2018-01-23:
am trying to write the whole program in java .... and i get "NaN" in the output of azel[0] , when i traceback the problem i found that 
        mk = eph.m0 + eph.n*tk;
	ek = mk;
	ekold = ek + 1.0;
               
        OneMinusecosE = 0; // Suppress the uninitialized warning.
	while(Math.abs(ek-ekold)>1.0e-14)
	{
		ekold = ek;
		OneMinusecosE = 1.0-eph.ecc*Math.cos(ekold);
		ek = ek + (mk-ekold+eph.ecc*Math.sin(ekold))/OneMinusecosE;
        }     

ek and ekold are the same even after adding +1 to ekold .... i printed out the result of both ek and ekold and they are the same ... that means am not able to go in to the while loop (which creates a problem because OneMinusecosE is 0) 

     ekdot = eph.n/OneMinusecosE;      this will be error , if the program didnt go through the while loop ...

osqzss, 2018-01-23:
There is no way that ek and ekold have the same value at the start of the while loop. The code must go into it.

robellebor, 2018-01-23:
so why do we need a while loop ... if "(ek-ekold)>1.0e-14" is always true when will the loop end ??

osqzss, 2018-01-23:
Kepler's equation must be solved for the eccentric anomaly, but it has no algebraic solution. The numerical solution can be obtained by an iteration method. You can find more details in any orbital mechanics book.

closed 2018-01-24 by robellebor