osqzss/gps-sdr-sim #133: receiving gps signal

bikilay, 2018-03-06:
from gps sdr simulation, i could transmitting using USRP N200.  it is transmitting well but when i tried to receive using GPS Data application on my phone, i faced a problem. i could not receive the signals even if able to transmit. ...how can i probably fix?

[image]

osqzss, 2018-03-06:
The default sampling frequency of gps-sdr-sim is 2.6MHz. If you want to transfer the I/Q samples at 2.5MHz, the sampling frequency of the gps-sdr-sim has to be set to the same value with '-s' option.

bikilay, 2018-03-06:
ok when am using 2.6MHz, it says like this:
  UHD Warning:
    The hardware does not support the requested TX sample rate:
    Target sample rate: 2.600000 MSps
    Actual sample rate: 2.631579 MSps
Press Enter to quit: 

[image]

HansHoogerwerf, 2018-03-06:
> HackRF and bladeRF require 2.6 MHz sample rate, while the USRP2 requires 2.5 MHz (an even integral decimator of 100 MHz).

 Maybe you want to try 2.5?

osqzss, 2018-03-06:
I mean '-s' option of gps-sdr-sim.

alkuna, 2018-03-07:
Besides the correct sampling frequency, I see two other possible problems:

1. Is 5-min record sufficient to provide full navigation and time messages?
2. Are you sure your phone is not confused when receiving GPS signal with date and time 4 years in the past?

Could you please try to download more recent brdc file and generate a file with 30 minutes of data?

closed 2018-03-21 by osqzss