osqzss/gps-sdr-sim #138: gps signal strength

robellebor, 2018-03-14:
I can receive the signals using usrp x310 ... but it's not fixing a 3D position because the satellite signals are weak and they change rapidly ... I tried increasing the gain but still, i cant get a lock...

Mictronics, 2018-03-16:
I had the same issue with the ADALM-Pluto that is using a 12bit DAC. Actually I got an improvement by using 9-bit sin/cos tables generating the baseband signal.
During experiments with gain, offset and bitshift values I noticed that you cannot simply increase gain. First this will decrease SNR, second, at some point there will be overflows in math that completely disturb your baseband signal. So use small steps when you make changes.

I found a good indication for signal quality when using the app SatStat on Android. When I see PRNs showing up in the SBAS, GLONASS and BEIDOU range than I know the signal is bad.

osqzss, 2018-03-16:
@robellebor
The "gain" parameter is used to create a "relative" signal power difference among GPS satellites up to about 10dB.  If you want to improve the "absolute" signal power, please increase the gain (or reduce the attenuation) of your signal transmitter.

closed 2018-03-21 by osqzss