osqzss/gps-sdr-sim #139: About delta timing improvement

Mictronics, 2018-03-16:
Not an issue actually, sorry for hijacking :wink: 

I'm working on code that is generating live IQ data for the ADALM-Pluto. Trying to improve the output I was thinking about the two lines below:

2169 computeCodePhase(&chan[i], rho, 0.1);

2343 grx = incGpsTime(grx, 0.1);

My understanding is that the calculations are performed with a fix delta time of 100ms independent of the real delta that my live IQ generation loop needs.
@osqzss 
Do you think of any improvement when I feed the measured real delta time into this functions?

osqzss, 2018-03-16:
What do you mean by "fix" and "real" delta times? What is the difference between them?

Mictronics, 2018-03-17:
By "fix" I mean the 0.1 function parameter when calling computeCodePhase and incGpsTime.
"real" would mean that this parameter is not a fix 0.1 but the delta measured between two function calls.
Background is that I use two threads to feed the ADALM-Pluto. The main thread is doing GPS math and IQ generation while a background thread is feeding the Pluto from IQ sample buffer.

However, I tried already with "real" delta measurements but no success or improvement seen.

osqzss, 2018-03-17:
You don't need to control the delta time in the code. What you need is a FIFO between the signal generation and transmission threads because these two are asynchronous. You first have to generate a certain length of I/Q samples in the signal generation thread and write them into the FIFO. While it generating the next period of data, the signal transmission thread read the I/Q samples from the FIFO and transfer them to the Pluto. The "real" transmission timing has to be managed by the Pluto side.

Mictronics, 2018-03-17:
> What you need is a FIFO between the signal generation and transmission threads because these two are asynchronous.
That's all done in the background by libiio. My code is already working. Was just wondering whether some timing needs to be taking into account. Obviously not.
Thanks anyway.

closed 2018-03-17 by Mictronics