osqzss/gps-sdr-sim #141: Ephemeris and subframes update

Mictronics, 2018-03-20:
`            // Refresh ephemeris and subframes
            // Quick and dirty fix. Need more elegant way.
            for (sv = 0; sv < MAX_SAT; sv++) {
                if (eph[ieph + 1][sv].vflg == true) {
                    dt = subGpsTime(eph[ieph + 1][sv].toc, grx);
                    if (dt < SECONDS_IN_HOUR) {
                        ieph++;

                        for (i = 0; i < MAX_CHAN; i++) {
                            // Generate new subframes if allocated
                            if (chan[i].prn != 0)
                                eph2sbf(eph[ieph][chan[i].prn - 1], ionoutc, chan[i].sbf);
                        }
                    }
                    break;
                }
            }
`
In the above code, can you explain what kind of "elegant way" is needed, maybe I can chime in with some help.
Also, why is the ephemeris data updated only within one hour? Related to the fact that ephemeris data changes only at hour boundaries? After one hour the code sends always same ephemeris data?

Lot's of questions I know, but want to understand the code some more. ;)
Thanks.

osqzss, 2018-03-21:
It's been a long time since I wrote this comment, and I don't remember what I was going to do. I guess I wanted to run this portion of code in another thread. Each GPS satellite broadcasts its own orbit parameters repeatedly every half-minute, and they won't change for an hour or so.

IvanKor, 2018-03-21:
void eph2sbf(gpstime_t g, const ephem_t eph, const ionoutc_t ionoutc, unsigned long sbf[MAX_SUBFR][N_DWRD_SBF])
{
	unsigned long wn;
	unsigned long toe;
	unsigned long toc;
	unsigned long iode;
	unsigned long iodc;
	long deltan;
	long cuc;
	long cus;
	long cic;
	long cis;
	long crc;
	long crs;
	unsigned long ecc;
	unsigned long sqrta;
	long m0;
	long omg0;
	long inc0;
	long aop;
	long omgdot;
	long idot;
	long af0;
	long af1;
	long af2;
	long tgd;

	unsigned long ura = 0UL;
	unsigned long code = 1UL;
	unsigned long svh = 0UL;
	unsigned long flag = 0UL;
	unsigned long fit = 0UL;
	unsigned long dataId = 1UL;
	unsigned long anti_spoof_flag = 0x8UL;
	unsigned long sbf4_page25_svId = 63UL;
	unsigned long sbf5_page25_svId = 51UL;

	unsigned long wna;
	unsigned long toa;

	signed long alpha0, alpha1, alpha2, alpha3;
	signed long beta0, beta1, beta2, beta3;
	signed long A0, A1;
	signed long dtls, dtlsf;
	unsigned long tot, wnt, wnlsf, dn;
	unsigned long sbf4_page18_svId = 56UL;
	unsigned long TLMWord = 0x8B0000UL;

	wn = (unsigned long)(g.week % 1024); // "transmission" week number FIX

	code = (unsigned long)(eph.code);
	svh = (unsigned long)(eph.svh);
	//svh = 0UL;
	flag = (unsigned long)(eph.flag);
	fit = (unsigned long)(eph.fit);
	toe = (unsigned long)(eph.toe.sec / 16.0);
	toc = (unsigned long)(eph.toc.sec / 16.0);
	iode = (unsigned long)(eph.iode);
	iodc = (unsigned long)(eph.iodc);
	deltan = (long)round(eph.deltan / POW2_M43 / PI);
	cuc = (long)round(eph.cuc / POW2_M29);
	cus = (long)round(eph.cus / POW2_M29);
	cic = (long)round(eph.cic / POW2_M29);
	cis = (long)round(eph.cis / POW2_M29);
	crc = (long)round(eph.crc / POW2_M5);
	crs = (long)round(eph.crs / POW2_M5);
	ecc = (unsigned long)round(eph.ecc / POW2_M33);
	sqrta = (unsigned long)round(eph.sqrta / POW2_M19);
	m0 = (long)round(eph.m0 / POW2_M31 / PI);
	omg0 = (long)round(eph.omg0 / POW2_M31 / PI);
[... 132 lines cut ...]
	end
	*/
	// Subframe 5, page 25
	sbf[4][0] = (TLMWord | ((wn & 0x3FFUL) << 2)) << 6;
	sbf[4][1] = (0x5UL | anti_spoof_flag) << 8;
	sat_sbf5_alm[0][0] = sbf[4][2] = (dataId << 28) | (sbf5_page25_svId << 22) | ((toa & 0xFFUL) << 14) | ((wna & 0xFFUL) << 6);
	sat_sbf5_alm[0][1] = sbf[4][3] = 0UL;
	sat_sbf5_alm[0][2] = sbf[4][4] = 0UL;
	sat_sbf5_alm[0][3] = sbf[4][5] = 0UL;
	sat_sbf5_alm[0][4] = sbf[4][6] = 0UL;
	sat_sbf5_alm[0][5] = sbf[4][7] = 0UL;
	sat_sbf5_alm[0][6] = sbf[4][8] = 0UL;
	sat_sbf5_alm[0][7] = sbf[4][9] = 0UL;
	sat_sbf5_alm[0][8] = 0xAAAAAA ;

	for (int m = 1; m < 25; m++){
		if (sat_sbf5_alm[m][8] != 0xAAAAAA){
			for (int j = 0; j < 8; j++){
				sat_sbf5_alm[m][j] = 0xAAAAAA << 6;
			}
		}
	}

	return;
}

IvanKor, 2018-03-21:
@osqzss, 
sorry,
this is the answer   @Mictronics

Mictronics, 2018-03-21:
This function updates the subframes. My questions were about the one hour update limit, referring to https://www.gps.gov/technical/ps/1995-SPS-signal-specification.pdf
Chapter 2.4.4.2 has the following sentence below table 2-5:
```
Cutovers to new data sets will occur only on hour boundaries except for the first
data set of a new upload. The first data set may be cut-in (reference paragraph 2.4.1.1) at any
time during the hour and therefore may be transmitted by the satellite for less than one hour.
```
I wonder if that is related to `dt < SECONDS_IN_HOUR`.

IvanKor, 2018-03-21:
>  My questions were about the one hour update limit

In[ RINEX files ](https://github.com/osqzss/gps-sdr-sim/blob/master/brdc3540.14n).

If you synthesize ephemeris, you need to know the algorithm for forming IODE.
IODE has a secret generation algorithm.

IvanKor, 2018-03-21:
> Not a big secret though. 

However, this is not described for the public.

> we can not calculate Ephemeris data on our own. 

[no problem](https://github.com/IvanKor/hrfGPS)  [ using almanac](https://github.com/IvanKor/hrfGPS/blob/master/screenshot2.png)

osqzss, 2018-03-21:
@Mictronics
The broadcast ephemeris is generally valid for two hours, one hour "before" and another hour after the time of ephemeris (TOE) or time of clock (TOC). This is why the simulator update the ephemeris when the difference between the current simulation time and TOC gets less than one hour. 

You can use the hourly prediction orbit or even alamanac as IvanKor mentioned. However, since the simulator uses the same orbit information for satellite location computation and navigation message generation, there is no range and position errors on the receiver side due to the satellite orbit prediction. So there is no "accurate" ephemeris for the simulator.

closed 2018-03-22 by Mictronics