osqzss/gps-sdr-sim #142: gps position not fixing

bikilay, 2018-03-21:
we are getting the gps signals with a  good strength but we can't get a fixed position and our guess is that the satellites are not spreaded as they should be.....
[image]
[image]
[image]
[image]

so why are the satellites in the same position ... real gps signals are spreaded when we recieve in our phone ...

IvanKor, 2018-03-21:
good strength  > 42

osqzss, 2018-03-21:
You couldn't get proper satellite location because the receiver failed to decode the navigation message. The signal strength seems rather low. You can increase it by 10dB. Typical C/N0 value under the open sky should be somewhere around 40 to 50dBHz.

bikilay, 2018-03-21:
we r transmitting the .bin file using usrp x310 ... and we set the usrp output gain to the maximum 31 dB (we even tried by makin it 100 dB) but still same output ... so where should we increase the 10dB gain ????

osqzss, 2018-03-21:
What kind of TX antenna are you using?

bikilay, 2018-03-21:
VERT2450 Vertical Antenna (2.4-2.5 and 4.9-5.9 GHz) Dualband

IvanKor, 2018-03-21:
most likely usrp is overloaded,  set the usrp output gain to the min

IvanKor, 2018-03-21:
Antenna (2.4-2.5 and 4.9-5.9 GHz) however use 1.75 GHz

osqzss, 2018-03-21:
Please test your GPS signal transmission with a receiver connected through a coaxial cable. Do not spread any unauthorized signal on the air.

kasmf, 2018-03-23:
still now all signal comes as the above image on the same position (on 0 degree). how to fix this problem 
in order to spread to different positions just the real gps signal and position fix?

Fles42, 2018-04-06:
Most of gps receivers have minimum elevation mask 5°...Maybe you can change it