osqzss/gps-sdr-sim #144: Unable to generate GPS signal file. getting "command not found"

apjcc, 2018-04-05:
I get "gps-sdr-sim: command not found" when running static gps signal generation command. Here is my log:

$ git --version
git version 2.7.4
$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/5/lto-wrapper
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 5.4.0-6ubuntu1~16.04.9' --with-bugurl=file:///usr/share/doc/gcc-5/README.Bugs --enable-languages=c,ada,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-5 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-5-amd64/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-5-amd64 --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-5-amd64 --with-arch-directory=amd64 --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.9) 
$ git clone https://github.com/osqzss/gps-sdr-sim.git
Cloning into 'gps-sdr-sim'...
remote: Counting objects: 547, done.
remote: Compressing objects: 100% (35/35), done.
remote: Total 547 (delta 39), reused 61 (delta 36), pack-reused 476
Receiving objects: 100% (547/547), 4.48 MiB | 2.72 MiB/s, done.
Resolving deltas: 100% (288/288), done.
Checking connectivity... done.
$ cd gps-sdr-sim
$ gcc gpssim.c -lm -O3 -o gps-sdr-sim
$ gps-sdr-sim -e brdc3540.14n -l 30.286502,120.032669,100
gps-sdr-sim: command not found
$

osqzss, 2018-04-05:
Try "./gps-sdr-sim" instead.

apjcc, 2018-04-05:
It worked. thanks.

closed 2018-04-05 by apjcc