osqzss/gps-sdr-sim #148: Output File Size Limit

TheStDrone, 2018-05-18:
Good afternoon,
I'm trying to generate a bin file with a duration of 2.5 hours. I've got the space to accommodate such a large file and I've changed the USER_MOTION_SIZE macro to what I need it to be, however, when I run the program, it only generates enough data for a ~826 second duration. This produces a file 4.3 GB in size, which is around the unique ID limit for a 32-bit system. I'm running a 64-bit system however, so if I recompile the program it should reallocate regular variables (such as int, long int, short int, etc.) as the 64-bit equivalent. Is there some restrictions in one of the variable declaration that is limiting the IQ data to be no more than 4.3GB worth of data?

Thanks

osqzss, 2018-05-18:
What is the file system format of your drive? You cannot store files larger than 4GB on a FAT32 volume. Formatting it as NTFS could resolve the issue.

TheStDrone, 2018-05-21:
Thanks for you help! That did seem to be the problem and reformatting the drive works!

closed 2018-05-21 by TheStDrone