osqzss/gps-sdr-sim #151: Cannot run without doppler shift

dlynge, 2018-06-13:
I am trying to simulate gps signals specifically without a doppler shift. How would I achieve this? So far, I've tried editing/commenting out parts of lines 1324-1328 in gpssim.c, but my attempts either don't produce a valid gps signal or the doppler remains shifted. Any help would be greatly appreciated.

Thanks.

osqzss, 2018-06-13:
How did you test the validity of the simulated GPS signal?

dlynge, 2018-06-14:
I use a ublox receiver and I calculate what the doppler shift should be (given the satellite's ephemeris) and what the measured doppler actually is. The code uses the ublox libraries and can precisely predict what the doppler should be.

When spoofing the satellites, I am well indoors, so I am unable to get real satellite signals. The problem that I'm having is that I can't get the measured spoofed signals' doppler to be different than the predicted values.

osqzss, 2018-06-14:
The Doppler measurements contain the receiver clock frequency offset. You can't compensate for it without calculating the velocity of the antenna (and the receiver clock frequency) from the Doppler measurements.

dlynge, 2018-06-14:
Ah, so that's what the offset is. I am not compensating for the receiver clock frequency offset, but it remains relatively constant so I can still look at trends between the calculated vs measured. Think of it as being precise, but not accurate.  Also, I'm running my tests stationary, so I know the velocity is zero.

closed 2023-05-10 by osqzss