osqzss/gps-sdr-sim #155: underrun in usrp n210

karanjarich, 2018-06-23:
Am using windows 8.1 i have gnu radio companion after sucessful installation ofusrp on my host computer i decided to run the "gps-sdr-sim-uhd.py -t gpssim.bin -s 2500000 -x 0" am getting an undean with UUUUUUUUUUUUUUUUU how can i prevent this and does it affecting signal transmission and does it affect signal transmission

karanjarich, 2018-06-27:
> use uhd_tx_samples_from_file if you are using usrp on windows.
> like: ./tx_samples_from_file --args="master_clock_rate=50e6" --file
> gpssim.bin --type short --rate 2500000 --freq 1575420000 --gain 0 --repeat
> refer: #50 <https://github.com/osqzss/gps-sdr-sim/issues/50>
>
> good luck!

i tried this way but unfortunately the cmd responds by saying is not
recognized as an internal or external command,
operable program or batch file.

osqzss, 2018-06-27:
It should be in the /usr/local/lib/uhd/examples/ folder.
https://kb.ettus.com/Verifying_the_Operation_of_the_USRP_Using_UHD_and_GNU_Radio

karanjarich, 2018-07-05:
this is how I solved the issue just benchmarked the usrp TX and RX using  \benchmark_rate --rx_rate 10e6 --tx_rate 10e6 currently i cannot see the flowgraph during transmission

chrsengel, 2019-01-11:
@karanjarich so what's the command that worked for you?