osqzss/gps-sdr-sim #159: no fix even though it can simulate GPS satellites.

ho0-kim, 2018-08-06:
Hello. I have a problem about 'fix location'.

With bladeRF and gps-sdr-sim, it was successful to generate fake GPS signals as you can see below:

[image]
[image]

But, the problem is location couldn't be fixed on my phone.

I turned off 'improve accuracy' and locating method was 'phone only'.

I've been stucked with the issue for 2 weeks but I couldn't find the reason...

Any advice will be appriciated.

ho0-kim, 2018-08-13:
I spoofed gps on other devices such as fitness bands.

But, I couldn't find the reason why it failed to spoof my phone yet.
I think older version of android is better for gps spoofing. Mine is android 8.

closed 2018-08-13 by ho0-kim

Honky103, 2018-08-13:
I also find it harder to spoof phones as compared to other GPS devices. One possibility could be the time difference between the time gotten via GPS and the internet. Never did find the actual reason though.

bikilay, 2018-10-22:
what did you use for spoofing? I mean usrp or hackRF..I am using USRP x310 for transmission of signal and use my android phone. I have installed gps test app and using it to spoof the position.  But i coudnt. i got the signal but cant fix position yet. I set my phone as phone only. made airplane mode. and turn off wifi and mobile data......I hope the problem is on .bin file while generating. anyone who can help me on USRP??

ho0-kim, 2018-10-22:
Hello, Solan.

I used Blade-RF for spoofing.
I think the problem is not on .bin file becuase you could check the gps
signal on your smart phone.
In my case, it takes time to fix location. The first fixation after running
gps sdr sim took around one minute.

atchyuth-rao, 2021-03-30:
hi,

i have generated GPS Signal for both static and Dynamic(user motion) coordinates with Pluto SDR, i am getting position fix for both cases with U-Blox Receiver.

but when i testing same scenario with mobile i am not getting even single satellite(no satellite C/N ratio bars are there)

can you brief me what should i do or follow for getting position fix with Mobile Phone

it would be great helpful to me

yosefzh, 2024-07-23:
> what did you use for spoofing? I mean usrp or hackRF..I am using USRP x310 for transmission of signal and use my android phone. I have installed gps test app and using it to spoof the position. But [...]

im having the exact same issue im having all satellites signals bars but i can't fix in any one of them for some reason using x310 and android phone for receiving