osqzss/gps-sdr-sim #161: limeplayer playing too fast

maximevince, 2018-08-13:
When playing a 1-bit 1 MSps file (or anything else) using limeplayer,
on a LimeSDR mini, it seems the file is being played back way too fast.

E.g. I am generating a 300 seconds file using:
`./gps-sdr-sim gps-sdr-sim -e brdc3540.14n -l 30.286502,120.032669,100 -b 1 -s 1000000 -d 300`

And playing it back using:
`./limeplayer -s 1000000.0 -b 1 -d 2047 -g 0.1  < ../gpssim.bin`

The payback streams data to my LimeSDR mini @ 185 MB/s,
and stops after ~3.5 seconds of playback:

```
Using device index 0 [LimeSDR Mini, media=USB 3.0, module=FT601, addr=24607:1027, serial=xxx]
Using normalized gain 0.100000
Reference clock 40.00 MHz
Using channel 0
Set sample rate to 1000000.000000 ...
actualRate 999999.990066 (Host) / 31999999.682109 (RF)
Calibrating ...
Tx Calibration: MCU error 5 (Loopback signal weak: not connected/insufficient gain?)
calibrate=22(Tx Calibration: MCU error 5 (Loopback signal weak: not connected/insufficient gain?))
Setup TX stream ...
1-bit mode: using dynamic=2047
gettimeofday()=> 1534168697:163755 ; TX rate:0.000000 MB/s
[...]
gettimeofday()=> 1534168702:010328 ; TX rate:183.885824 MB/s
gettimeofday()=> 1534168702:026381 ; TX rate:183.885824 MB/s
./limeplayer -s 1000000.0 -b 1 -d 2047 -g 0.1 < ../gpssim.bin  2.93s user 3.11s system 65% cpu 9.270 total

```

I remember having using this tool to correctly playback on this very same LimeSDR in the past,
So I am wondering what has changes, or what I am doing wrong.

osqzss, 2018-08-14:
I don't have a mini, but this may fix your trouble:
https://github.com/osqzss/LimeGPS/issues/2

maximevince, 2018-08-15:
Thanks for the pointer.
I have tried it; and while it does change behavior,
it does not fix the problem.
I wonder what has changed, since this was working before...

What TX rate should I expect for a 1bit, 1 Msps file?

Example: -s 1000000 -> sample rate 1 Msps -> is not accepted anymore by the LimeSDR driver:
```
device[1/1]=LimeSDR Mini, media=USB 3, module=FT601, serial=1D40EC59CAF526, index=0
Using device index 0 [LimeSDR Mini, media=USB 3, module=FT601, serial=1D40EC59CAF526, index=0]
Using normalized gain 1.000000
Reference clock 40.00 MHz
Using channel 0
TX0 Channel has 3 antenna(ae)
Channel 0, antenna [NONE] has BW [0.000000 .. 0.000000] (step 0.000000)
Channel 0, antenna [BAND1] has BW [2000000000.000000 .. 2600000000.000000] (step 0.000000)
Channel 0, antenna [BAND2] has BW [30000000.000000 .. 1900000000.000000] (step 0.000000)
sampleRateRange [0.100000 MHz.. 30.720000 MHz] (step=0.000000 Hz)
Set sample rate to 1000000.000000 ...
SetFrequencyCGEN(4 MHz) - cannot deliver requested frequency
setSampleRate=-1(SetFrequencyCGEN(4 MHz) - cannot deliver requested frequency)
```

maximevince, 2018-08-15:
It seems it was working on: LimeSuite  Library version:	v18.03.0-ge4a1abbe
It does not work on:   Library version:	v18.06.1-g81fb282e,
which I am now running...

Keep you posted

maximevince, 2018-08-23:
I have tracked down the issue to be introduced since LimeSuite git revision 52d61291f789c7bba070da2642be93aa9dcd6621

https://github.com/myriadrf/LimeSuite/commit/52d61291f789c7bba070da2642be93aa9dcd6621

Can you make anything out of that?
Which software is probably the root cause?

maximevince, 2018-08-23:
Found a workaround for limeplayer:
Enabling an RX channel, will initialize it correctly, causing the TX stream to follow the right sample rate (~3 MB/sec. Will file a bug @ LimeSuite and provide a pull request for a workaround over here.

closed 2018-10-01 by osqzss

jeremyherbert, 2019-03-21:
I realise this issue is closed, but thank you @maximevince for posting the commands you used. I now have this working using a LimeSDR Mini thanks to you!