osqzss/gps-sdr-sim #163: Simulating Real Device

karanjarich, 2018-08-16:
Hey, Guys hope you are all well.

I'm thinking of how to simulate real device such as the behavior of GPS speed limiter on a given route. My pseudocode is as follows
     plot path on google map and speed at given points
     generate nmea 
my question is that i want the ground speed interpolated on nmea to affect the signal generated for sdr to emulate the real device. How can achieve this i already have JS to create different nmea files

osqzss, 2018-08-16:
Try SatGen and nmea2um.
https://forums.hak5.org/topic/38290-gps-simulator/

IvanKor, 2018-08-17:
https://github.com/IvanKor/hrfGPS

karanjarich, 2018-08-17:
thanks, @osqzss I have used both but my goal is at the end of the day I will test device after defining the route, a specific speed at different places within the route, how they will behave during specific spoofing. I have even created my nmea generator that captures speed and altitude but how can interpolate speed during .bin generation