osqzss/gps-sdr-sim #165: Windows 10

mgbx, 2018-09-05:
Hi, 
I was just wondering if this is easy to use on wondows 10 or should I use Ubuntu?

osqzss, 2018-09-05:
I'm using Windows, but it should work on Ubuntu.

mgbx, 2018-09-05:
I'm having issues using it after I build it in Visual Studio. It just shuts down. Tried looking at a previous thread regarding something similar, but couldn't solve the issue. 

Any pointers/hints? We're using your software along with a HackRF One in a bachelors assignment, trying to spoof a DJI drone. But as you may understand, getting nowhere without the program working.

osqzss, 2018-09-05:
How did you run the software? You have to run it on a command prompt window to specify options.

osqzss, 2018-09-08:
I'm having no trouble to build the software with VS2017 and run it on Windows 10:
[https://drive.google.com/open?id=1lZPTYtHcYtvITYiZwDhNymXUCMFiYv4G](https://drive.google.com/open?id=1lZPTYtHcYtvITYiZwDhNymXUCMFiYv4G)

You have to open the command prompt first and type in the command to run the software.

mgbx, 2018-09-09:
Thank you, of course it was user error from my side. 
We are pretty green to this, would you be able to give us some advice on setting up for right ephemerides and maybe some other things so this could be a little more plug 'n play?

Much appreciated

osqzss, 2018-09-10:
Please try the prebuilt executable in the release page:
https://github.com/osqzss/gps-sdr-sim/releases

Open a command prompt windows from the start menu, move to the release folder, and type in the following command line:

`> gps-sdr-sim -e brdc3540.14n`

mgbx, 2018-09-26:
Ok, so far it seems to work. My next issue is really to transfer it to the hackrf. 
"> hackrf_transfer -t gpssim.bin -f 1575420000 -s 2600000 -a 1 -x 0"
This command isn't recognized as an internal or external command.

Sorry for disturbing you with this, but I'm really trying to figure this out.

osqzss, 2018-09-26:
You first have to build the HackRF host tools.
https://github.com/mossmann/hackrf/tree/master/host

mgbx, 2018-10-01:
Okey. At this time, we can't seem to install the HackRF host tools, mostly because of lack of competence I guess, but we're not able to get the installation to work properly.

Do you have any suggestions/advice for what we can do from here, since we may not be able to test the gps-sim in our bachelor? Any help is much appreciated.

osqzss, 2018-10-01:
Try PothosSDR. It contains the prebuilt HackRF host tools (and many others) for Windows. 
https://github.com/pothosware/PothosSDR/wiki/Tutorial#download-and-install

You can find the hackrf_transfer executable and all the necessary dependencies in the PothosSDR/bin folder.

mgbx, 2018-10-03:
Thank you, it seems to be working fine now. May be questions that occur later though.

mgbx, 2018-10-10:
I have found recent brdc file from NASA. But it doesn't seem to change the location of anything, not our drone or phones or gps-watches. What things are we missing to determine a new position to broadcast from the SDR?

mgbx, 2018-10-24:
Hi,

We seem to have a found out why the drone won't change position. It doesn't seem like the HackRF is transmitting anything, despite the TX led glowing. Any ideas?

komalkumar6655, 2019-09-20:
hello sir , I am new to visual studio can you please explain me how to build GPS sdr sim in windows.

osqzss, 2019-09-20:
1. Start Visual Studio.
2. Create an empty project for a console application.
3. On the Solution Explorer at right, add "gpssim.c" and "getopt.c" to the Souce Files folder.
4. Select "Release" in Solution Configurations drop-down list.
5. Build the solution.

komalkumar6655, 2019-09-20:
Can we also generate that GPS signal using cmd prompt??