osqzss/gps-sdr-sim #170: Iphone 6

th2112, 2018-09-22:
Has anyone had any luck with getting this to work on an Iphone 6?  I am trying to test some location apps using different static locations on an apple 6 phone. I'm having no luck. I can get this to work on several Android phones and tablets, but not with the Iphone.  I have tried the latest brdc file combined with the current date and time. I have no sim in the phone, and have tried it with the wifi on and off. Below is the options I am trying.

./gps-sdr-sim -e current_brdc -l 42.214603,-80.363839,100 -b 8 -t $(date -u +%Y/%m/%d,%H:%M:%S) -o myoutput.bin

hackrf_transfer -t myoutput.bin -f 1575420000 -s 2600000 -a 1 -x 26 -R

FYI I am using a hackrf with a external tcxo.  Google maps (on my android seems to work just fine ) 

Any Ideas?

-- below is a very crude sh script to delete any current brdc files is a few directories and replace them with the newest brdc file from the ftp site.  In case that helps anyone. 

#!/bin/sh
day=$(date +%j)
year=$(date +%Y)
yr=$(date +%y)
site="ftp://cddis.gsfc.nasa.gov/gnss/data/daily/$year""/brdc/""brdc""$day""0.$yr""n.Z" 

cd "/home/pi"
rm "current_brdc"
cd "/bin"
rm "current_brdc"
cd "/home/pi/gps-sdr-sim"
rm "current_brdc"
cd "/home/pi"

wget "$site"
uncompress "brdc""$day""0.$yr""n.Z"
echo epheremis "brdc""$day""0.$yr""n.Z"
mv "brdc""$day""0.$yr""n" "current_brdc"
cp "current_brdc" /home/pi/gps-sdr-sim
cp "current_brdc" "/home/pi/bin"

TachyonZZ, 2018-09-23:
If the phone is on a cell network it's positioning off of tower information as well. If the iPhone is receiving live sky then it's likely using GLONASS as well as GPS, and if it has a data connection it's using AGPS information. This will only work reliably if the AGPS state is cleared (and not updated over the air) and you're not competing with other constellations.  If the device is in a GNSS denied environment simply clearing the AGPS state and disabling any other location enhancements like WiFi should make it work.

TachyonZZ, 2018-09-23:
I'm not sure how to clear the agps on an iphone... The only ios device I've used this simulator with is an ipad and had no issues so long as it is denied live sky signals.  With android devices an app called "GPS Status" has an option to clear the agps. The app is available for ios but I can't speak to whether it has the same function. 

As for constellations, it's always going to digest both GPS and GLONASS signals if available. If denied both, it should be tracking your simulated GPS signals. Where problems can arise is when the receiver is tracking live sky signals. Many receivers made in the last 4-5 years couple GLONASS and GPS so if you inject one that disagrees with the other it may not produce a position fix. 

If I recall correctly the time I used this with the ipad it did take considerably longer to start tracking as compared to other devices, but eventually it did produce a fix.

th2112, 2018-09-24:
thanks, that was the key, I had rf leakage into my room and all I was conserned about was GPS I was not checking for other constellations. It was enough to get a fix and thus would not pickup on my simulated gps.
Thanks!

closed 2018-09-26 by th2112