osqzss/gps-sdr-sim #179: gps-sdr-sim kills gps, but no signal

elduda, 2018-11-29:
Hi guys,
I'm trying to let my car "think" he is in the US. Instead of Europe.
Bought the hackrf and some other stuff in the end, because it didn't work yet.

So here are the commands I use:
make
./gps-sdr-sim -b 8 -e brdc3330.18n -l 40.6860224,-74.3188546,100
hackrf_transfer -t gpssim.bin -f 1575420000 -s 2600000 -a 1 -x 0

all the comands work.
I also tried to put a 10mHz square wave to the CLK IN - red that in some forum. But nu sucess yet. The hack rf is only killing all the gps datas, but no mobile phone finds the new gps datas.

Any ideas or other information you'd need?

Best
David

osqzss, 2018-12-04:
You may need extra attenuation between the HackRF and the ANT500. If the signal is too strong, it's simply jamming the receiver.

Mictronics, 2018-12-08:
The guaranteed minimum signal power on earth (receiver side) is -127dBm to -136dBm. Compare this to the output levels of hackrf. Modern GPS receiver may include a logic to detect high signal levels as jamming and stops operation. Or the receivers front end gets simply RF saturated.

elduda, 2018-12-08:
Thanks a lot!
whats the best solution to reduce the signal? is there a way via commands? or do I need hardware extensions?

simonhook4, 2019-02-20:
I currently have the same problem.  Is there any fix yet?  I have used different brdc files but still no signal.

Simon

osqzss, 2019-02-20:
HackRF requires a good external 10MHz clock, preferably <2ppm.

simonhook4, 2019-02-23:
Thank you. I’ll add one and see how it goes.

simonhook4, 2019-02-28:
> HackRF requires a good external 10MHz clock, preferably <2ppm.

ok i now have one and installed it.  How do i go about getting it to work with the hackrf?

osqzss, 2019-02-28:
This would be helpful: https://forums.hak5.org/topic/38290-gps-simulator/

simonhook4, 2019-04-07:
im not having much luck here.  I have 50dB attenuation connected to the ANN500 Ariel.

I also have the new clock, have checked it is working by running hackrf_debug --si5351c -n 0 -r and have the correct output.   

Am i missing something?  I do block out the gps on my phone as it goes grey and says it can't find a gps signal

simonhook4, 2019-04-14:
Success. I decided to format and reinstall everything under a clean system.  

1 thing i did change was lower the run time to 60 second and use -R to repeat the signal. 

My main reasons for that was because I’m using a raspberry and it takes much longer to create the binary file. 

Thank you

simonhook4, 2019-05-13:
Does anyone know why some phones work quickly and some not at all from the same device?