osqzss/gps-sdr-sim #180: A question about dwrd

hash-hash, 2018-12-18:
Hello. I have another qustion. Why the array dwrd contains 60 elements instead of 50?

osqzss, 2018-12-20:
The simulator needs to generate the signal transmitted earlier than the reception time. The first 10 dwrd elements contains the previous subframe 5 contents.

hash-hash, 2018-12-20:
Thank you for your answer! 
Theoretically, could it start the simulation from a previous subframe 4 or 3 (then dwrd would have 70 or 80 elements) instead of subframe 5?

osqzss, 2018-12-23:
Yes, you can start simulation from any GPS signal reception time with -T option. Since the gps-sdr-sim updates the whole subframe contents every 30 seconds at the end of subframe 5, you only need a copy of the previous subframe 5.

terrymarkovich, 2021-04-22:
Another question related to this. It appears that subframe 1 transmission starts with dwrd 9 bit 26 and ends with dwrd 59 bit 25. Does this mean that the first few bits of subframe 1 and the last few of subframe 5 are missed?

terrymarkovich, 2021-04-23:
I got it. dwrd 9 and 59 are identical, so nothing is missed.

venkat-512, 2024-10-27:
But  isn't it enough if the 5th subframe is sent only the first time  of the total duration? If we do it everytime while we send the navigation message is sent - aren't we sending like 5-1-2-3-4-5,5-1-2-3-4-5,5-1-2-3-4-5...etc .Wouldn't this be wrong ? Great work with the code by the way .