osqzss/gps-sdr-sim #183: How are "observable" satellites determined?

TheGregor, 2019-01-01:
It is my understanding that you can input a static location (Lat / Long / Alt) and a start time (hh:mm:ss) and, using an ephemeris file, generate signals for the satellites that should be visible from that location at that time.  However, when I do so I receive data for an entirely different set of satellites than those that were actually observed at the time.  The information is correct, insofar as I can generate a PVT fix that matches the target location, just not using the satellites that were observed in real time.  Any ideas as to what could cause this disparity?  I've verified that the inputs to gps-sdr-sim are correct, and that the correct ephemeris is being used.

TheGregor, 2019-01-05:
Yes, I was using the -t / -T flag, and between "These files are then used to generate the simulated pseudorange and Doppler for the GPS satellites in view" and  "The simulation start time can be specified if the corresponding set of ephemerides is available.", propagating satellite orbits is exactly what I expected.  Is there any way to do such a thing with this program?  I would prefer to create replicas of specific moments with specific satellites rather than simply spoofing locations.

Thanks!

osqzss, 2019-01-05:
The ephemeris parameters are only valid for a couple of hours. If you can obtain the latest ephemeris from the live GPS signal, use "-t" (lower case letter) option. It will propagate the satellite orbit to the specified time. I personally recommend to use UBX outputs from a u-blox receiver to obtain the current broadcast ephemeris.

chrsengel, 2019-01-11:
What is the correct format for the `-T` option?