osqzss/gps-sdr-sim #185: One plus 6T can't be deceived

K1two2, 2019-01-14:
One plus 6T can't deceive, I turned on the flight mode and always showed "GPS positioning".

[image]

[image]

osqzss, 2019-02-26:
They are getting smarter :)

closed 2019-03-02 by K1two2

kitwou, 2019-03-25:
@osqzss is there any way we can lock these devices with spoofed signals? like jamming BDS and other GNSS? i have noticed some newer devices locks with our GPS when we switch to airplane mode and reboot (no data connection after reboot), can you explain in detail if possible what measures should i take to make these devices lock?
hardware used: bladeRF, hackrf, USRP - B210.

dkdhruv123, 2019-05-13:
can you tell me on what basis these phones ignore our spoofed signals? or what can i do to make these phones lock to mine signal? jamming each and every BDS and glonass? transmit at more power ? 
one thing i have noticed that when i switch on airplane mode and reboot (no cellular connection after reboot), these phones get locked within 1 minute but not with cellular/wifi turned on. any solutions ?

Shangu-xsg, 2021-04-21:
Hello, can you tell me what SDR you use to complete GPS spoofing? Also, do you have an external antenna on the SDR? During my GPS spoofing, the signal I received from the GPS was always too weak.