osqzss/gps-sdr-sim #186: Input data from external GPS receiver (question)

uzhas-sovka, 2019-02-21:
I want to offset my current position 10 kilometers to north. SDR board doesn't have GPS receiver, so I obtain GPS position from Arduino-controlled GPS chip hanging on a long wire 10 meters from SDR board. 

My goal is to 1) get data from Arduino, 2) add 10 kilometers to latitude, 3) transmit spoofed signal from SDR.

Please give me some advice, I don't know where to start. This project seems to help upload already compiled program to SDR chip, and I can't understand how can I read external information by SDR board.

osqzss, 2019-02-26:
gps-sdr-sim is an off-line I/Q signal generator. You can't control the simulated GPS signal in real-time.

uzhas-sovka, 2019-02-27:
Thanks. I have found this thread
https://github.com/osqzss/gps-sdr-sim/issues/73

closed 2019-02-27 by uzhas-sovka