osqzss/gps-sdr-sim #187: realtime pipe

patvdleer, 2019-02-26:
Since I am able to generate the `.bin` faster than the runtime, would it be possible to generate the data in real time and pipe it directly through to `hackrf_transfer`?

osqzss, 2019-02-26:
I'm not a Linux person, but I think you can use namedpipe.

$mkfifo  named_file 
$./gps-sdr-sim -e brdc3540.14n -u circle.csv  -b 8 -o named_file
$hackrf_transfer -t named_file   -f 1575420000 -s 2600000 -a 1 -x 0
$rm named_file

patvdleer, 2019-02-26:
This wouldn't change the problem I think. I've been looking into the code but I'm not C guy, I would like to be able to realtime edit the current lat,long, i.e. hook it up to a controller and control it

osqzss, 2019-02-26:
There seems several real-time versions available: https://github.com/osqzss/gps-sdr-sim/issues/73