osqzss/gps-sdr-sim #189: Carrier Phase

alexrosatisw, 2019-04-08:
Hello,
I'm new here. I'm trying to understand how this code works, but I'm experiencing some difficulties with line 1625. I was wondering: why, to calculate the carrier phase, is phase_ini multiplied by 512 and by 65536?

https://github.com/osqzss/gps-sdr-sim/blob/1506e5414f855159df7316b63343cc5c325bd8b0/gpssim.c#L1625

Thanks for your time, have a nice day!

osqzss, 2019-04-08:
Floating vs. fixed point arithmetic. 1.0 floating carrier cycle = 16-bit (65536) integer. Then the single carrier cycle is divided into 9-bit (512) phase for sin/cos lookup tables.

alexrosatisw, 2019-04-09:
Thanks for your clear and quick answer,
I wish you a pleasant day!