osqzss/gps-sdr-sim #190: Alternate rinex source

ongbuntu, 2019-04-14:
Hi guys,

I found an alternate rinex source here:
ftp://igs.bkg.bund.de/NTRIP/BRDC/

However, I am unable to use it in the same way as those downloaded from cddis. I think some of the header information is different.

Is anyone able to advice if the rinex files can be somehow converted to be compatible?

Any advice is deeply appreciated.

Meown12, 2019-05-29:
Hi, 
if you compare the headers of the rinex files from your source and for example ftp://cddis.nasa.gov/gnss/data/daily/2019/brdc/ you notice, that the files from the bund.de server are missing the ion alpha + beta, delta utc and leapsecond lines, which IIRC are necessary for the initalization of the gps generation.
Source: CDDIS:
```
     2              NAVIGATION DATA                         RINEX VERSION / TYPE
CCRINEXN V1.6.0 UX  CDDIS               29-MAY-19 03:26     PGM / RUN BY / DATE 
IGS BROADCAST EPHEMERIS FILE                                COMMENT             
    0.6519D-08  0.2235D-07 -0.5960D-07 -0.1192D-06          ION ALPHA           
    0.8602D+05  0.9830D+05 -0.6554D+05 -0.5243D+06          ION BETA            
    0.279396772385D-08 0.710542735760D-14   503808     2055 DELTA-UTC: A0,A1,T,W
    18                                                      LEAP SECONDS        
                                                            END OF HEADER 
```
Source BUND:
```
     2.11           Navigation data                         RINEX VERSION / TYPE
BNC 2.12.8          rtproc              29-May-19 07:19     PGM / RUN BY / DATE
Source: RTCM_3.2 www.igs-ip.net/ABMF00GLP0                  COMMENT
//EDIT: many many more source lines (118 lines)
Source: RTCM_3.3 www.igs-ip.net/ZIM200CHE0                  COMMENT
                                                            END OF HEADER
```
That could be the issue at hand.

Best luck

Max

[brdc-comparison.zip](https://github.com/osqzss/gps-sdr-sim/files/3231247/brdc-comparison.zip)

Mictronics, 2019-05-29:
A version that might work is located in:
ftp://igs.bkg.bund.de//GREF/BRDC/
But is missing leap seconds.

ongbuntu, 2020-03-15:
Thanks for all the replies. Just wanted to officially close this.

closed 2020-03-15 by ongbuntu