osqzss/gps-sdr-sim #191: compiling gps-sdr-sim-realtime using usrp X310

bikilay, 2019-04-19:
when i try to compile with "$ gcc gpssim.c -lm -O3 -o gps-sdr-sim" compiler on ubuntu using usrp X310, the following error generated.....the error is pointing in to "socket.c" which is about connecting to TCP.......

$ gcc gpssim.c -lm -O3 -o gps-sdr-sim
In file included from gpssim.c:14:0:
socket.c: In function ‘timem’:
socket.c:41:2: warning: implicit declaration of function ‘gettimeofday’ [-Wimplicit-function-declaration]
gettimeofday(&t, NULL);
^~~~~~~~~~~~
socket.c: In function ‘udprecv’:
socket.c:60:41: warning: passing argument 6 of ‘recvfrom’ makes pointer from integer without a cast [-Wint-conversion]
return recvfrom(s, dataa, siz, 0,&from,sizeof(from));
^~~~~~
In file included from /usr/include/x86_64-linux-gnu/sys/socket.h:269:0,
from socket.c:2,
from gpssim.c:14:
/usr/include/x86_64-linux-gnu/bits/socket2.h:64:1: note: expected ‘socklen_t * restrict {aka unsigned int * restrict}’ but argument is of type ‘long unsigned int’
recvfrom (int __fd, void *__restrict __buf, size_t __n, int __flags,
^~~~~~~~
In file included from gpssim.c:14:0:
socket.c: At top level:
socket.c:63:30: warning: excess elements in array initializer
double llhr[3]={39.68,139,76,10};
^~
socket.c:63:30: note: (near initialization for ‘llhr’)
gpssim.c: In function ‘usage’:
gpssim.c:1651:9: warning: format ‘%f’ expects argument of type ‘double’, but argument 2 has type ‘char *’ [-Wformat=]
printf("Usage: gps-sdr-sim [options]\n"
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
gpssim.c:1665:3:
" -n Use TCP connect to Gnuradio TCP-Source for\n realtime simulation.\n",

@bikilay

osqzss, 2019-04-19:
Which source code did you try to compile? It seem not the one in the master repository.

bitdust, 2019-04-19:
for usrp, you can try my fork： https://github.com/bitdust/gps-sdr-sim

bikilay, 2019-04-22:
I am using the source code from  gps-sdr-sim-realtime.  I used gps-sdr-sim-master and able to lock or fix position by generating .bin file and transmitting it over USRP......but here i could only use downloaded brdc file instead of real satellite signal....know I am trying to spoof position by using real signal which comes directly from satellite outdoor...

osqzss, 2019-04-22:
u-blox receiver provides raw navigation message stream in hex format. Read the current ephemeris from the stream and feed it to the simulator.

ferorted, 2019-12-13:
Hi. Anyone can explain how to feed the simulator from ublox receiver output?
Can I get the message stream in hex format from receiver usb port?
then,  so how feed this stream to gps-sdr-sim-realtime?

Thank you

bitdust, 2019-12-14:
you can use u-center software to extract & save the the ephemeris data.
or use rtklib to receive from ublox and parse the RAW output to get the ephemeris file.

ferorted, 2019-12-14:
Thanks for thenswer.
However, I really want to simulate a static location on the fly.
I mean, receiving raw ephemeris and Nav messages from usb port of Ublox receiver, modifying it with gps-sdr-sim (to obtain a preset static location) on the fly and feed the SRD.
Is it possible?

bitdust, 2019-12-14:
as far as i know, gps-sdr-sim doesnt have a 'stream mode' to produce signal stream~
so you cant manipulate messages 'on the fly'.
i tried use gnss-sdr and gps-sdr-sim to do this job but failed :
`orignal signal` --> `GNSS-SDR` --> `message stream` --> `modify data` ---> `gps-sdr-sim` 
because it needs too much coding works.

FayeWong98, 2023-10-12:
This is a question worth discussing. 
Has anyone implemented this function? 
I look forward to more discussions.