osqzss/gps-sdr-sim #192: Doppler shft, need to add a "blue" offset

IvanKor, 2019-05-02:
chan->f_carr = -rhorate / LAMBDA_L1 **- 15.9**

Then much more accurate

P.S.
However there are still minor inaccuracies

lenhart, 2021-01-04:
Hi Ivan,

first: I am no maintainer of this project, just curious :-)

how did you derive your correction parameter for the doppler shift? (Calculation/Simulation/Testing)
I think it would be very beneficial to add this in a comment if adopted in the source code. Magic numbers are always confusing and future improvements on the number will be easier if one knows how it was derived :-)
Cheers!

IvanKor, 2021-01-04:
using u-blox N8T GPS receiver to receive see "pseudorange residual" in u-center

precise correction : **-15.7542 Hz**

#define DELTA_DOPPLER_FREQ (-15.7542)
chan->f_carr = -rhorate / LAMBDA_L1 + DELTA_DOPPLER_FREQ ;

lenhart, 2021-01-07:
Hi Ivan, 
strangely I no longer see your reply to this thread, but thanks anyways!
I think the relevant part of your answer was:

`using u-blox N8T GPS receiver to receive see "pseudorange residual" in u-center`

but then you came up with a much different value for being more precise. I think I'll check the value myself next time I use u-center, I just wondered if it maybe isn't time stable?

IvanKor, 2021-01-07:
> strangely I no longer see your reply to this thread,

the exact value "blue" offset is the top secret of the USA
so i deleted the message

> I just wondered if it maybe isn't time stable?

for HackRF I use external generator
http://mxtal.ru/ru/products/?S=76&C=89&I=201
have better temperature stability and phase noise compared to the Rb standard.
and very precise frequency setting

such generator is three times more expensive than HackRF

Mictronics, 2021-01-07:
@lenhart 
I upgraded my hackrf's with an OCXO DOC020F-010.0M.
https://www.digikey.de/product-detail/de/connor-winfield/DOC020F-010.0M/CW722-1-ND/2619615
This one has +-20ppb (1ppb = 1/10^9 = 10^-9) but is available down to +-1.5ppb. Just depends on what your spendings.

Mictronics, 2021-01-07:
> the exact value "blue" offset is the top secret of the USA
> so i deleted the message
You don't want to bit the hand that feeds you, right?

IvanKor, 2021-01-07:
> > the exact value "blue" offset is the top secret of the USA
> > so i deleted the message
> 
> You don't want to bit the hand that feeds you, right?

following  top secret  

https://github.com/osqzss/gps-sdr-sim/issues/270

> I generated and transmitted the signal with HackRf. My recorded range for successful spoofing is 2 feet max. Is there a way to improve the range? **I have come across that there is a 20db attenuator in the code.** Can we disable it for improving the range?