osqzss/gps-sdr-sim #193: The first seconds don't register location?

xNotta, 2019-05-15:
Not sure if this is the right place to ask this, but I don't know where else to go.

I've been successful at creating and simulating gps routes with the HackRF.
Everything is working but there's been a problem where the first 30-60 seconds of my route doesn't register then it instantly snaps to location. It's kind of difficult to explain.

My route consist of points in this order: A, B, C, D, E, F, G, A
But when I run the simulated gps, it starts at D after some seconds then continues though E, F, G, and back to A then stops for 30+ seconds as the file repeats. After those seconds pass, I end up back at D and run though the order again. So it never works with point A, B, C at the beginning of the file.
A solution was to leave the locations the same and just put in the expected route after point D but the run time is exceeding 600 seconds as compared to the original 360 seconds.

I have tried different location for all point but it's always the first less than 60 seconds that don't work.

Using SatGenNemaFree, I set my max speedto 30km/h, Stationary period and stationary period end to 1 (originally 80 and 5) in order to cut off 85 seconds.

I am using a KML that was converted using "SatGenNMEAFree" then converted to User Motion using /satgen/nema2um then simulated using gps-sdr-sim and transmitted though a HackRF.

Any idea why this is happening?

simonhook4, 2019-05-19:
My guess is that is because the phone is sorting itself out.  If you do a static mode simulate then it also takes that amount of time to register.  

My suggestion would be to create the first point at A for over a minute so the device can then get adjusted to the signal you send out.

xNotta, 2019-05-27:
Thank you for the reply. That makes sense. I've been experimenting with the "Stationary period" and "Stationary period end" but it still doesn't seem to be helping those first few locations. 

> My suggestion would be to create the first point at A for over a minute so the device can then get adjusted to the signal you send out.

How would I do this properly? Is there a better NMEA software to use other than SatGenNMEAFree?