osqzss/gps-sdr-sim #194: Beidou signal

lk815160423, 2019-06-03:
Have you considered implementing a simulation of the Beidou signal in your program?
I tried it and found that the D1 and D2 signal formats of the Beidou signal are very complicated.

osqzss, 2019-06-21:
Not for the moment. Any contribution is welcome :)

whysoft, 2021-12-05:
I have an idea.
Can you record the BeiDou or Gps signal in one place and play it in another place.
For example, you go to New York first and record the local signal. 
Later, if you want to spoof location of New York in Paris, play this sampling signal file.

superbongbong, 2023-10-31:
May I ask if you were successful in your attempt to simulate a Beidou signal? I've been working on a similar project recently

cehuidong, 2024-07-24:
> 请问您模拟北斗信号的尝试是否成功？我最近一直在做一个类似的项目

May I ask if you were successful in your attempt to simulate a Beidou B1I signal? I want to modulate a BeiDou B1I signal with reference to a modulation similar to a GPS signal, and when I verify the signal using GNSS-SDR it shows that the software can capture the satellite but cannot lock on to it.