osqzss/gps-sdr-sim #195: Segmentation fault (core dumped) on newer ephemeris file.

r00tb3, 2019-06-19:
I've downloaded the latest 2019 ephemeris file from ftp://cddis.gsfc.nasa.gov/gnss/data/daily/
after that, I followed the following steps...

`gcc gpssim.c -lm -O3 -o gps-sdr-sim -DUSER_MOTION_SIZE=4000`

```
gps-sdr-sim -e brdc0930.19n -u circle.csv -b 8
Start time = 2019/04/03,00:00:00 (2047:259200)
Duration = 300.0 [sec]
10  232.4  51.4  21186373.3   3.0
12   57.1  54.8  20897108.1   3.1
14  312.2  31.4  22970795.4   3.9
15  125.2   9.0  25062308.9   8.9
20  199.8  32.8  22488568.1   4.2
24   64.2  25.7  23187100.3   5.6
25  158.5  74.7  20148093.5   2.6
29  146.1   9.9  24714004.9   8.2
31  266.6  19.9  23467499.4   4.5
32  318.0  46.8  21602325.9   3.1
Time into run = 300.0
Done!
Process time = 57.7 [sec]
```

```
gps-sdr-sim -e brdc0930.19n -g triumphv3.txt -b 8
Segmentation fault (core dumped)
```

```
gps-sdr-sim -e brdc0930.19n -l 30.286502,120.032669,100 -b 8
Using static location mode.
Start time =    0/00/00,00:00:00 (0:0)
Duration = 400.0 [sec]
ERROR: No current set of ephemerides has been found.
```
Then I compiled the same with the old ephemeris file(brdc3540.14n) it worked properly and generated **gps-sdr-sim** file. 

When I transmit using` hackrf_transfer -t gpssim.bin -f 1575420000 -s 2600000 -a 1 -x 0 ` it transmits data but my smartphone's GPS location isn't spoofed nor am I receiving strong signals when placed closed to [HackRF](https://i.imgur.com/sE3M8oZ.png?1). 

What am I missing? Where am I going wrong?

osqzss, 2019-06-21:
If you're using a HackRF, this might solve the issue: https://github.com/osqzss/gps-sdr-sim/issues/198

closed 2019-06-22 by r00tb3

Antanide, 2021-03-01:
This actually happens _before_ sending any data to the hackrf, it's gps-sdr- sim that segfaults. This is still a current problem.
Thanks,

Mictronics, 2021-03-01:
@Antanide Please attach a brdc file that causes a segfault.