osqzss/gps-sdr-sim #198: Receiving signals but co-rodinates unchanged.

r00tb3, 2019-06-20:
My smartphone is receiving signals but the location is undetected.

[image]
[image]

What might be the problem?

@osqzss

Johsnon, 2019-06-20:
Hi,
what SDR device is this? Hackrf? Is there an additional TCXO (=clock) installed?
[image]
I had the same problem, because the clock of my hackrf wasn't precise enough.

r00tb3, 2019-06-22:
> Hi,
> what SDR device is this? Hackrf? Is there an additional TCXO (=clock) installed?
> [image]
> I had the same problem when using hackrf without that clock part (the internal clock of my hackrf wasn't precise enough).

It's HackRF and I'm not using an external clock with the HackRF.

Where can I buy the external clock that you've mounted on the HackRF?

Also, will I need a passive GPS antenna?

osqzss, 2019-06-22:
You can make one by yourself ;)
https://github.com/osqzss/gps-sdr-sim/tree/master/extclk

osqzss, 2019-06-22:
A simple dipole antenna works just fine.
https://www.u-blox.com/sites/default/files/products/documents/GPS-Antenna_AppNote_%28GPS-X-08014%29.pdf

r00tb3, 2019-06-22:
Following this [setup](https://github.com/osqzss/gps-sdr-sim/blob/master/extclk/hackrf_gpssim.jpg) I'm buying a [passive GPS antenna](http://www.digikey.com/product-search/en?keywords=TS.07.0113), [RF attenuator](https://www.mouser.in/ProductDetail/HUBER%2bSUHNER/6630_SMA-50-2-199_NE?qs=sGAEpiMZZMvo7grblV%252BRTALl1QIn%252BZZTdTmTBD6cXrygPAKU7ITuMQ%3D%3D) & [TCXO Module](https://www.nooelec.com/store/tiny-tcxo.html).

Also, do I need the 30db attenuator as shown [here](https://github.com/osqzss/gps-sdr-sim/blob/master/extclk/hackrf_gpssim.jpg)?

What if I don't use the attenuator?

osqzss, 2019-06-22:
It will disturb the GPS receivers in your neighborhood.

r00tb3, 2019-06-23:
> Hi,
> what SDR device is this? Hackrf? Is there an additional TCXO (=clock) installed?
> [image]
> I had the same problem when using hackrf without that clock part (the internal clock of my hackrf wasn't precise enough).

Did you also change the antenna? Or just used provided an external clock of 10Mhz and it worked?

kozlovsas, 2019-07-04:
I'm having a similar issue where my car's GPS is receiving the signal but is not changing location. I'm using a HackRF with a TXCO module from NooElec, and an ANT500 in a controlled environment. The same occurs with my smartphone- I've turned off Wifi/Cellular/Bluetooth and when I launch Google Maps my location is not changed. I'm just broadcasting a static location, should I broadcast something else?

ogord, 2019-07-04:
Hi, I have the same problem, but with LimeSDR. I use the next parameters:
./gps-sdr-sim -e brdc1850.19n -s 2500000 -l 35.170514,25.009579,100 -b 8 -o sim111.bin
./limeplayer -s 2500000 -b 8 -g 0.5 < ../sim111.bin 
Device configuration:
os ubuntu 16.04
LimeUtil --info
 Version information:
  Library version:	v19.04.0-gd20557de
  Build timestamp:	2019-07-04
  Interface version:	v2019.1.0
  Binary interface:	19.04-1
SoapySDRUtil --info
Lib Version: v0.8.0-gfe8dfd18
API Version: v0.8.0
ABI Version: v0.8
Install root: /usr/local
Search path:  /usr/local/lib/SoapySDR/modules0.8
Module found: /usr/local/lib/SoapySDR/modules0.8/libLMS7Support.so (19.04.0-d20557d)
Available factories... lime
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8

r00tb3, 2019-07-04:
> I'm having a similar issue where my car's GPS is receiving the signal but is not changing location. I'm using a HackRF with a TXCO module from NooElec, and an ANT500 in a controlled environment. The [...]

I guess you'll ahve to get a GPS antenna I doubt if ANT500 will work!!

osqzss, 2019-07-04:
@r00tb3 
The signal strength shown in your picture seems too low. They should be somewhere around 45dB.
By the way, the nominal TX power of HackRF or any other SDR devices is very high. You can use ANT500 to transmit the GPS signal without any trouble.

r00tb3, 2019-07-10:
I've mounted the TXCO module and a GPS antenna so now the spoofing attack works!!

closed 2019-07-10 by r00tb3

kozlovsas, 2019-07-10:
Thanks @osqzss for confirming the ANT 500 will indeed work, I figured that made no sense why it wouldn't. I'll open a separate issue for my case.