osqzss/gps-sdr-sim #199: gps sdr

EzanaA, 2019-06-21:
Hello, sir thanks for your valuable work on the gps sdr sim project. I am trying to simulate the project and everything is fine so far but when the position is fixed once after a few seconds went off and ON (Fixed and not Fixed) by itself. The signals coming are strong enough but not stable, come and go within a few seconds. even if i could fix the position, i will not stay as fixed. it looses fix within a fraction of time. what will be the problem expected? am using USRP x310.
Thanks in advanced

osqzss, 2019-06-21:
What is your target device?

djshivaana, 2021-06-24:
Hi @EzanaA  I am also trying to simulate gps on USRP x310 but i am facing undarrun problem, Can you please help me how to resolve this issue.
Thanks in advance