osqzss/gps-sdr-sim #200: problem of not

bikilay, 2019-06-21:
hello over there
I am working on Gps sdr-simulation..I could fix the position but he signal coming is not stable not stable. Even if the signals coming are strong enough, it took more than ten minutes for first time fix, and after fix it looses the position within a fraction of seconds.  Again took from 5 to 10 minutes to fix again......and so on......what will be the problem?? How could I fix it?? any one who can help...
   I am using USRP x310 and VERT900 antenna. and while transmitting the signal over USRP, I used different level of gain from 0db to 30db.

osqzss, 2019-06-21:
What is the "duration" of your simulated signal file?

yangaoning, 2019-06-22:
hello,
The nearly same problem was found when I used the USRP N200 to simulate the GPS signal. It was not stable at all. Though I used the same file gpssim.bin, the results showing by the app of AndroiTS GPS were different. Sometimes the simulating signal was rightly calibrated and the location was given out nicely. But usually not.
Why? I had examined part of the coded and no problem was found. 
And now I think the reason is that the TXCO in the USRP is not stable because the accuracy of the TXCO is 2.5ppm which is bad for the simulating signal with the central frequency of 1575.42MHz.
Unfortunately the above reason may be seemed reasonable, but I have no way to verify. 
Maybe someone can help me. It is my great pleasure.

osqzss, 2019-06-22:
Generally, 2.5ppm is accurate enough for the simulator, and many GPS receivers are able to acquire the signal without any problem. What is your sampling frequency? There is a special requirement for the N200 clock: https://github.com/osqzss/gps-sdr-sim/issues/133

yangaoning, 2019-06-23:
hello,Osqzss.
Thank you very much!

My simulating conditions were as follows: the sampling rate was set to be ' -s 2500000', and the byte of the I/Q data was set to be ' -b 16'. The resulting gpssim.bin was 3.0GB. Sometimes my mobilephone was spoofed to the simulating location. Though the simulating signals were calibrated sometimes, after a few seconds it failed. Usually the simulating signal couldn't be expected.

The second question: if the accuracy of the TXCO is 2.5ppm, the frequency error of the TXCO with GPS L1 frequency equals 3.9KHz (2.5ppm*1575.42GHz=3.9KHz). The Doppler frequency may be 4.9KHz. When the TXCO frequency error was added to the Doppler frequency, could the mobilephone capture or track the simulating GPS signals easily, timely and correctly. I am thinking over this question.

Maybe I am writing too directly for you to enjoy reading the above quetions.
Best wishes! I'm grateful for your patience.

osqzss, 2019-06-25:
It requires a bit wider search range, but 2.5ppm frequency offset should not be any problem for typical GPS receivers.

bikilay, 2019-09-02:
Hello, I could fix my position by compiling Gcc and then generating GPS signal on terminal....But now i need to work it advanced way. instead of opening terminal, compile and generate signal, i try to develop user interface that enable user simply by pushing button that leads to compile and generate signal...in general compiling and generating signal like clicking the button let "GENERATE". when the button generate clicked it will generate signal as we have been doing on terminal....anybody who can help or give me a hint......I tried using PyQt environment writing my own code with python (GUI), as C and python can be compile together using cPython extension. i face problem on how to link C program or main code (gpssim.c) and my code (Python code).........

osqzss, 2019-09-02:
There seem to be several different ways to call a C executable from Python script:
https://www.quora.com/How-do-I-write-a-Python-script-to-run-a-C-program
https://stackoverflow.com/questions/2473655/how-to-make-a-call-to-an-executable-from-python-script

RannyTheCoder, 2020-07-13:
@yangaoning ... did you solve your issue ? can you pls explain how