osqzss/gps-sdr-sim #202: Possible doppler effect error high speeds

DL7AD, 2019-07-10:
Hi,

I'm trying to simulate high altitude situations. I used the rocket.csv and satellite.csv demos therefore. When the simulation is not moving everything works fine but as it gets drastically fast (like above 500m/s), my receiver loses GPS lock. It locks shortly as the rocket get slower as it reaches the top of ascent. Could there something be wrong with the doppler effect which is calculated according to those speeds? Simulating static locations doesn't pose a problem.

Sven

osqzss, 2019-07-10:
You need a very special receiver to overcome that speed limit.
https://space.stackexchange.com/questions/14687/current-situation-with-cocom-regulations-and-gps-receivers-for-balloons-and-cube

DL7AD, 2019-07-10:
I have that kind of receiver.

osqzss, 2019-07-10:
I've tested several different space-capable GPS receivers with my gps-sdr-sim demo scenarios but never had a similar trouble. Do you mind if I ask which GPS receiver you have?

DL7AD, 2019-07-30:
This was not an error by the gps simulator rather than the signal strength at the GPS receiver. There are still problems receiving it, but the cause lays at the receiver itself.

closed 2019-07-30 by DL7AD