osqzss/gps-sdr-sim #207: Problem with generating .bin file

alex512125, 2019-07-29:
Hello!I have tried to transmit some location with my HackRF but GPS Test at android phone even don t see any strong signals  ! I searching in other issues and found out that .bin works perfectly for me (https://github.com/osqzss/gps-sdr-sim/issues/33#issuecomment-189151493) .It fix location if 30 seconds.But when i generate .bin by gps-sdr-sim phone even dont see signals from HackRF!!

alex512125, 2019-07-29:
[image]
In sdr#
I use Ubuntu 19.04

osqzss, 2019-07-29:
Show us the command line that you executed.

alex512125, 2019-07-29:
hackrf_transfer -t test.bin -f 1575420000 -s 2600000 -a 1 -x 0

Thats your firmware from comment 33
https://drive.google.com/drive/folders/0B36BcNBXEPjzUmlxX1dYUVRsT00

osqzss, 2019-07-29:
I mean the command line options of gps-sdr-sim.

osqzss, 2019-07-29:
You need "-b 8" option to generate 8-bit I/Q stream for HackRF.

alex512125, 2019-07-29:
Thank you very much.Now it shows signals!!

closed 2019-07-29 by alex512125