osqzss/gps-sdr-sim #208: Wrong and ustable altitude HackRF One

ArinaVetrova, 2019-07-30:
Hello!
I have a problem using HackRf One with Ublox Lea-M8T and GEOS-5 GPS modules.

Command for generation *.bin file:
./gps-sdr-sim -b 8 -e brdc1130.19n -l 30.286502,120.032669,100 -s 1500000 -o coordinates.bin
coomand for start generation:
hackrf_transfer -f 1575420000 -s 1500000 -a 0 -x 5 -R -t coordinates.bin

Coordinates are correct but altitude is always wrong and ustable. It increases and decreases, can become negative(-113 m for example) of super-big (like 30000 m) and then become normal again. But it never stably shows 100 meters, it's always 150-200-300-50 and etc, slowly changes every few seconds. 
Also 3d fix is not stable too. It sometimes become "No fix" and back to "3d fix" .

I tried different level , sample frequency, file duration, coordinates, that didn't help.

This HackRf has external clock but I'm bot sure about it's PPM
[image]
[image]
[image]

.

osqzss, 2019-07-30:
The bandwidth of GPS C/A code is about 2MHz. You need to set a higher I/Q sampling rate than that.

ArinaVetrova, 2019-07-31:
Thanks for the answer. I tried with sample rate 2.3 MHz, altitude changes slower and range is not so big, but it still doesn't stable at 100 meters.

ArinaVetrova, 2019-08-09:
Hello guys. Seems like it's ublox's problem about wrong altitude. On another GEOS-5 altitude is 100 m.
Your advice try to increase sample rate was very helpful.

closed 2019-08-09 by ArinaVetrova

IvanKor, 2019-08-09:
А что отвечает (смотреть при помощи U-center) Ublox Lea-M8T на команду UBX-CFG-PT (продакт тест) 06 41 или в HEX B5 62 06 41 00 00 47 DB 
?

stef-stack, 2019-08-21:
Hello,

The same problem ;)

Command for generation *.bin file:
./gps-sdr-sim -b 8 -e brdc2240.19n -l 50.941609,6.959483,100  -o gpssim.bin
coomand for start generation:
hackrf_transfer -f 1575420000 -a 1 -x 5 -R -t gpssim.bin

I use a Odroid to create and transmit the signale. 

Coordinates are correct and i always become "3d fix" with U-Blox but altitude is always wrong and ustable.