osqzss/gps-sdr-sim #209: USRP B210 phone cant receive

EzioAuditore09, 2019-08-08:
Hi. I have used usrp b210 with -s 2500000 -b 16 -x 0. As there were no problems. USRP was transmitting the signal, I have checked with a spectrum analyzer. But my mobile phone (S6 edge) cannot receive any signals on GPS test application and does not fix the location(with use only gps option). I have tried with different gain values but the result is still the same. The signal was transmitted through adalm pluto as well  and was showing up in the spectrum analyzer but my phone still cant get the signal. (I have used current epherimis data) I think the problem is not related the sdr hardware. Best regards.

StefanoSperetta, 2020-06-25:
We use it on a daily basis with -s 2600000 and it works like a charm.

StefanoSperetta, 2020-06-25:
As part of a university course on Aerospace Engineering, we have students use regular GPS receivers and simulate trajectories to make them practice with real-life scenarios. Unfortunately with the COVID lockdown, going around with a GPS receiver and performing measurements was not really possible, so most of the assignments were simulated.

james-999, 2020-11-26:
you can try [https://github.com/kobekobe1234567/gps-sim-ui](url) it works fine for any cellphone. but you may need to uncheck "use current time" checkbox. i use a Chinese produced B210 and a 0.01PPM OCXO.

FuhrerKing33, 2021-07-13:
Greetings, i am facing a similar issue, have you found a fix yet?