osqzss/gps-sdr-sim #210: bash: ./gps-sdr-sim: No such file or directory

Torino-Uni, 2019-09-04:
hi guys please help me 
i did everything correct but once i am getting the same problem:::: :~/gps/gps-sdr-sim$ ./gps-sdr-sim -e brdc3540.14n -l 30.286502,120.032669,100
bash: ./gps-sdr-sim: No such file or directory

what is wrong in my case 
thank your help

AkifT53, 2019-09-04:
change ./gps-sdr-sim with ./gps-sdr-sim-uhd.py
Yo just forgot adding -uhd.py
The true one is below
gps-sdr-sim-uhd.py -t gpssim.bin -s 2500000 -x 0

AkifT53, 2019-09-04:
Hi
Did you do the steps below?
1. Download the zip file
2. Extract it
3. Open the extracted file
4. Open the terminal in extracted file
5. Write this: gcc gpssim.c -lm -O3 -o gps-sdr-sim
The gps-sdr-sim executable should be shown up in the folder instantaneously after these steps. Make sure you did the steps correctly.