osqzss/gps-sdr-sim #213: Help need with understanding the code.

ShahZahidKhan, 2019-10-19:
Hello. I am a research student. I have been able to successfully run the code for a spoofed static location and predefined trajectory. Now I need to tweak the code for the output power etc. In Order to do that I need to understand the Code and its working first. But Unfortunately, my programming skills are not up to par. Can anybody help me in understanding the code?  Maybe some tutorials, relevant materials or a commented version of the code.

Your help would be greatly appreciated.

komalkumar6655, 2019-10-28:
hello HI. I am a research student , I have downloaded the brdc file and it is so much long .
but the IF data generated is small .
can you please explain how the IF data is generated using the RINEX nav. file??

atchyuth-rao, 2021-03-09:
> Hello. I am a research student. I have been able to successfully run the code for a spoofed static location and predefined trajectory. Now I need to tweak the code for the output power etc. In Order [...]

which hardware are you using for testing