osqzss/gps-sdr-sim #214: Can't fix using hackrf

yisonyang, 2019-10-22:
I used hackrf one to test.The phone can't fix all the time. Gps Test APP shows that some satellite can be seen ,SNR is around 55 .But the elevation angle and azimuth angle of all the satellites are  always 0.
What's problem could I meet?

Antanide, 2019-10-30:
Did you try with different transmission powers and distances from the receiving unit?

yisonyang, 2019-10-31:
> Did you try with different transmission powers and distances from the receiving unit?

Thanks for your response sincerely, sir! Yes. I tried hackrf_transfer -x range from 0 to 47.I changed the antenna few days ago. Now it works on some old android phones but  doesn't get better on some new android phones .

xxxxxaaaaaaaannnn, 2020-03-08:
Hello, I posted #240 that might help you.

yisonyang, 2020-03-09:
> Hello, I posted #240 that might help you.

Thanks a lot for your post,Sir.I will test it.

RannyTheCoder, 2020-07-13:
did it work for you ?

yisonyang, 2020-07-16:
> did it work for you ?

No,I still don't figure out the problem.