osqzss/gps-sdr-sim #215: how IF data is generated using RINEX file

komalkumar6655, 2019-10-28:
hello HI. I am a research student , I have downloaded the brdc file and it is so much long .
but the IF data generated is small .
can you please explain how the IF data is generated using the RINEX nav. file?

osqzss, 2019-10-28:
The duration of the IF data is specified by the "-d" option, not the length of the BRDC file.

komalkumar6655, 2019-10-28:
can you please explain how IF data is generated using nav. file

komalkumar6655, 2019-11-05:
[image]

What does the White selected portion means 
please give a explation.
thankyou.

osqzss, 2019-11-05:
PRN, azimuth, elevation, range, and ionospheric delay of the visible satellites.

komalkumar6655, 2019-11-06:
how the user location is attached to the generated prns