osqzss/gps-sdr-sim #222: Using gps-sdr-sim with USRP E310

andersonreloaded, 2019-12-10:
Hi Folks,

I have had unsuccessful attempts in using gps-sdr-sim with USRP E310 platform.
As per the instructions initially gps signla file was generated using GPS broadcast ephemeris file.
Subsequently, data was played back using the python script gps-sdr-sim-uhd.py and handheld Garmin GPS receiver was used to test the reception. However, things didn't worked as expected and the receiver never acquired the signals.

Since the tool was tested with USRP2, are there any specific parameters to be modified in order to generate GPS signal file and run it on E310 platform. 
Suggestions please...

osqzss, 2019-12-12:
Did you try tx_samples_from_file tool?
https://kb.ettus.com/Verifying_the_Operation_of_the_USRP_Using_UHD_and_GNU_Radio

andersonreloaded, 2019-12-13:
Tried through tx_sample_from_file tool, but beyond sampling rates of 4M, it throws underflow error.
Same is the behavior with gps-sdr-sim python script.

bitdust, 2019-12-13:
the read throughput (from TF card) of E310 is too low~

andersonreloaded, 2019-12-13:
@bitdust : Then, is there a way out to do simulation of GPS transmission using E310 ?
I have no other SDR platform and need to test with limitations of E310.

bitdust, 2019-12-13:
tool `benchmark_rate` in `/lib/uhd/examples`

andersonreloaded, 2019-12-14:
Benchmark tool works well up to sampling rates of 8MS/s. For higher rates it results in underflow error.

andersonreloaded, 2019-12-18:
What should be the data type for generating I/Q file so that it can be played back with E310 ?

andersonreloaded, 2019-12-20:
Finally got the setup working! Hence closing this thread.

closed 2019-12-20 by andersonreloaded

cyrus104, 2020-04-30:
@andersonreloaded , I have an E312 and would like to see how you got this setup working?

I'm looking for a way to load it up, I haven't put anything on the USRP besides the base OS.

andersonreloaded, 2020-05-15:
@cyrus104 , currently I am caught up with certain important activities.
I need some time to active the USRP setup before I can share the required details.
Shall try my best to respond at the earliest.

cyrus104, 2020-06-04:
@andersonreloaded, understand priority comes up and thank you for trying to pull the configs off.