osqzss/gps-sdr-sim #225: Problems getting devices to lock using Adalm-Pluto

toppharley, 2019-12-26:
Hi,

I'm having some trouble to get any tested device to lock on simulated location, using Adalm-Pluto:

Linux pluto 4.14.0-41915-gc2041af #279 SMP PREEMPT Mon Jan 14 13:13:47 CET 2019 armv7l GNU/Linux Machine model: Analog Devices PlutoSDR Rev.B (Z7010/AD9363)

1)
File to be played:

pi@raspberrypi:~ $ ./gps-sdr-sim/gps-sdr-sim -e ./brdc.n -l 41.661636,-4.7446656,630 -t 2019/12/24,12:00:00 -o ./gpssim2.bin -s 10e6 -b 8 -v
Using static location mode.
8.382e-09 -1.490e-08 -5.960e-08 1.192e-07
9.421e+04 -1.311e+05 -1.311e+05 8.520e+05
3.72529029846e-09 6.21724893790e-15 405504 2085
18
Start time = 2019/12/24,12:00:00 (2085:216000)
Duration = 300.0 [sec]
02 221.6 30.5 22691094.9 3.7
05 304.7 66.5 20530759.3 2.2
06 185.6 9.3 24752200.0 6.5
07 53.0 46.3 21876248.5 2.8
09 79.7 11.2 24550751.8 5.9
13 274.8 39.4 21957357.0 3.0
15 274.5 7.6 24660240.1 5.2
28 141.6 15.4 23932301.2 5.8
30 103.4 75.9 20424508.4 2.1
Time into run = 30.0
02 221.4 30.3 22705012.0 3.7
05 304.5 66.8 20523720.9 2.2
06 185.5 9.1 24773209.5 6.6
07 52.9 46.1 21887898.0 2.8
09 79.9 11.1 24567389.4 5.9
13 275.0 39.6 21944882.8 3.0
15 274.6 7.8 24642381.9 5.2
28 141.4 15.6 23915663.7 5.7
30 102.4 75.9 20424895.2 2.1
Time into run = 60.0
02 221.3 30.1 22718988.0 3.7
05 304.3 67.0 20516768.1 2.2
06 185.5 8.9 24794211.0 6.6
07 52.8 45.9 21899612.3 2.8
09 80.0 10.9 24584032.7 5.9
13 275.2 39.7 21932449.8 2.9
15 274.8 7.9 24624533.9 5.1
28 141.3 15.8 23899048.3 5.7
30 101.4 75.8 20425381.8 2.1
Time into run = 63.8
^C

2)
Playing the file:
pi@raspberrypi:~ $ ./gps-sdr-sim/player/plutoplayer -t ./gpssim2.bin

Transmit file size: 1.187 GB
Acquiring IIO context
Found (Analog Devices Inc. PlutoSDR (ADALM-PLUTO))
Acquiring devices
Context has 5 device(s).
Acquiring TX device
Initializing streaming channels
Enabling IIO streaming channels
Creating TX buffer
Transmit starts...
Done.

3)
Taking a look
./retrogram-plutosdr --rate 5e6 --bw 3e6 --freq 1575.42e6 --step 0.01e6 --ref-lvl 0 --uri usb:1.7.5
( no profile change)

4)
Also, getting no locks either when using other software:
pi@raspberrypi:~ $ pluto-gps-sim/pluto-gps-sim -e ./brdc.n -B 3.0 -A -30 -l 41.661636,-4.7446656,0 -U usb:1.7.5 -t 2019/12/24,12:00:00 -v
Using static location mode.
8.382e-09 -1.490e-08 -5.960e-08 1.192e-07
9.421e+04 -1.311e+05 -1.311e+05 8.520e+05
3.72529029846e-09 6.21724893790e-15 405504 2085
18
Gain: -30.0dB
RINEX date = 24-DEC-19 06:42
Start time = 2019/12/24,12:00:00 (2085:216000)
PRN Az El Range Iono
02 221.6 30.5 22691415.0 3.7
05 304.7 66.5 20531337.2 2.2
06 185.6 9.3 24752301.5 6.5
07 53.0 46.3 21876703.7 2.8
09 79.7 11.2 24550874.3 5.9
13 274.8 39.4 21957756.8 3.0
15 274.5 7.6 24660323.6 5.2
28 141.6 15.4 23932468.9 5.8
30 103.4 75.9 20425119.5 2.1

I'm stumped,... any help would be greatly appreciated.

Cheers!

Mictronics, 2019-12-26:
I assume with device you mean a mobile. Are you aware that on modern mobile device GPS is only one of several services to obtain the device location. Also, real GPS has very low guaranteed signal strength everywhere on earth. Your device may include jamming detection by strong signals and therefore doesn't consider the Pluto signal being valid.

jpelayo, 2019-12-26:
I understand; but anyway, the ./retrogram-plutosdr  should detect a profile, but does not.

Narhog, 2020-01-14:
I've the same issue.
But then my attenuation is higher than -55.0 dB, the regular GPS-signals get jammed.

jamesl-dm, 2020-11-09:
Try turning off your pluto's xo_correction (fw_setenv xo_correction 40000000 on console), and use a baseband data rate of 4 MHz. Set the attenuation to 80 dB, and put the receiver within 30 cm of the pluto.

atchyuth-rao, 2021-03-09:
I am not getting a position fix with the PLUTO SDR card,  I have generated a signal with a -100dbm signal power using PLUTO SDR and fed to U-Blox8 GNSS Receiver, satelliteS are acquired but their C/N Ratios are changes gradually i.e. not constant C/N ratio, I am getting position fix and gone 

can you help me out to resolve the issue

jamesl-dm, 2021-03-09:
Try this .grc file in gnuradio companion.
You need to disable xo_correction on your Pluto, then tweak the value in the ppmOutTarget variable of the grc file.

[gps-sdr-sim.zip](https://github.com/osqzss/gps-sdr-sim/files/6105847/gps-sdr-sim.zip)

atchyuth-rao, 2021-03-09:
thank you so much for your response,

I am not using gnu radio, I am having plutoplayer.c code which is linux compatible, using this plutoplayer.c code I am generating RF signal by following the commands in gps-sdr-sim.zip file

please helpme out to resolve the issue

jamesl-dm, 2021-03-09:
I suggest you install gnuradio. Disabling xo_correction will probably be enough though, so long as your GPS receiver can tune in the remaining ppm on its own.

atchyuth-rao, 2021-03-09:
is there any other way to solve this problem?

Mictronics, 2021-03-09:
> fw_setenv xo_correction 40000000

This will not turn off XO correction but overwrites the factory calibration value of the oscillator.
Keep in mind:
The output frequency will be generated by multiplication from reference frequency (40000000 Hz). Any offset in reference will be multiplied at the output. Setting 4000000 Hz as (assumed) fix reference might create a difference to the real TCXO/OCXO frequency. This difference is then multiplied and a visible offset at 1575 MHz.

My two Pluto using 40000017 and 40000004 to create a precise output at 1575.42 MHz.

jamesl-dm, 2021-03-10:
The issue is, if you use the xo_correction the relationship between the RF center frequency and the baseband sample rate gets recalculated and quantized according to a complex algorithm in the Pluto driver. And if you're unlucky, the quantized values will be so far off that the baseband sample rate will slip in and out of phase coherence with what the GPS receiver is expecting based on the center frequency. This causes the signal strengths reported by the receiver to oscillate, and the receiver loses lock. So my solution is to set xo_correction to nominal, and account for the quantization and ppm correction in software. You even have to account for the gnuradio resampler block quantizing its arguments into a single precision float - the GPS receiver is that sensitive!

atchyuth-rao, 2021-03-10:
Hello James,

Earlier I have Generated GPS signal samples with 2.6MHz sampling frequency
using Pluto SDR, with that sampling frequency satellite C/N ratios are
oscillates very frequently in GNSS Receiver

Later I have changed sampling frequency to 2.5MHz with this sampling
frequency I am getting constant 2D fix in GNSS Receiver, I am not getting
3D fix

I have done nothing regarding xo_correction for Pluto  in GNU radio

jamesl-dm, 2021-03-10:
The .grc file I posted is configured for a 4 MHz sample rate, and the xo_correction set to 40000000.
That's all the help I can offer - I'm unsubscribing from this thread now.

atchyuth-rao, 2021-03-18:
Dear james and mictronics,

i have modified xo_correction to 40MHz of pluto sdr in config.txt file
which was given by pluto SDR mass storage

[image: image.png]

GPS IQ Samples Generation:
[image: image.png]

U-blox GNSS receiver Results:
[image: image.png]
i am getting constant 2D fix, i am not getting 3D fix , can you help me out
where is the issue.

atchyuth-rao, 2021-03-18:
[image]
[image]
[image]
[image]

Mictronics, 2021-03-18:
In u-center open message view (F9), check if your receiver supports UBX-RXM-MEASX message. In case yes, check that MP is indicated low or medium for the simulated satellites. When MP is high increase the attenuation (-a option). You can also check the UBX-MON-HW message. Noise level should be as low as possible, same for CW jamming indicator. AGC somewhere in the middle.

What is your setup? Is that a direct cable connection or via antenna?

atchyuth-rao, 2021-03-18:
Thanks for your reply,

Pluto SDR TX is connected to Ublox receiver with -90dbm attentuation
(-50dbm as Pluto attentuation and -40 external attentuator)

I have Generated 12 satellites out of 12 I am getting atmost 5-6 only and I
am not getting 3D-fix

Is there any hardaware/configurations modifications still I need to do
further, I have generated GPS IQ streams from GPS-SDR-SIM with 2.5Msps
sampling rate.

Please share your views for getting proper 3D fix with Pluto SDR

Mictronics, 2021-03-18:
[image]
[image]
Code: https://github.com/Mictronics/multi-sdr-gps-sim
Setup PlutoSdr (with OCXO) -> 10dB attenuator -> ublox M8T
Commandline: `gps-sim -g-80 -f -l -12.0,-16.0,400 -e rinex2.gz -snow -v --radio=plutosdr`

Ephemeris and almanac for all satellites
[image]

atchyuth-rao, 2021-03-19:
Thanks for your response.

I will try with above mentioned procedure

Only concern is for me I have done nothing regarding TCXO modification of
Pluto SDR.

atchyuth-rao, 2021-03-20:
Dear mictronics,

I have followed the procedure as you mentioned in the mail, my pluto SDR
device is not recognized, please go through the below screenshots once.

gps-sim Executable_Creation:
[image: image.png]

running command:
[image: image.png]

TUI terminal Screenshots:

[image: image.png]
[image: image.png]
[image: image.png]
[image: image.png]

what should i modify please let me knoww

atchyuth-rao, 2021-03-20:
small modification,

i am using this command:

 gps-sim -g-80 -f -l -12.0,-16.0,400 -e rinex2.gz -snow -v --radio=plutosdr

Mictronics, 2021-03-20:
Images are not shown.

atchyuth-rao, 2021-03-20:
I have attached screenshots in an earlier mail, for your reference I am
sending again

atchyuth-rao, 2021-03-20:
can you help me out  faster, i have tucked with PLUTO SDR since 15-20 days

A little bit of help from you is awaited

Mictronics, 2021-03-20:
```
gps-sim Executable_Creation:
[image: image.png]

running command:
[image: image.png]

TUI terminal Screenshots:

[image: image.png]
[image: image.png]
[image: image.png]
[image: image.png]
```
No images....

> can you help me out  faster, i have tucked with PLUTO SDR since 15-20 days
Everyone wants everything faster. Complain when you work 15-20 month on this.

atchyuth-rao, 2021-03-20:
Thanks for your response

What I mean to say, I have stucked since 15days, if you guide/help me I
will finish it off my Project faster

Don't think in any other way, I will be having so much gratitude towards
you

I have attached files to you but I didn't know why it is not showing to you

I will send you again

atchyuth-rao, 2021-03-20:
PFA,

atchyuth-rao, 2021-03-22:
hi Micronics,

have you received Images??

Mictronics, 2021-03-22:
Received? No. What's the problem with attaching them to this PR? Simple copy and past or upload works for me. Maybe you stop quote reply.

atchyuth-rao, 2021-03-22:
No I am sending again for you...but I don't know what's happening, it's ok
again I will send you

Mictronics, 2021-03-22:
You just reply to notification Emails from Github, right? Did you check where that goes? I guess you can send your images a hundred times, they will not care. m(

atchyuth-rao, 2021-03-22:
How should I send images to you?

I have no idea where it's gone wrong,

I need your help to debug 3D fix issue with Pluto SDR

atchyuth-rao, 2021-03-22:
[image]

atchyuth-rao, 2021-03-22:
[image]

atchyuth-rao, 2021-03-22:
[image]

atchyuth-rao, 2021-03-22:
[image]

Mictronics, 2021-03-22:
It seems this is way over your head. Try a more simple project...
You compiled the code without PlutoSDR support. The README says:
 
> With ADLAM-PLUTO support: `make all PLUTOSDR=yes` (depends on libiio and libad9361-iio)

You need to build and install library dependencies before if not already done.

atchyuth-rao, 2021-03-22:
Yes I already run that command, but my Pluto SDR is not connected

atchyuth-rao, 2021-03-27:
hi mictronics, i am getting position fix, i have uninstalled all librararies and follow your Multi-SDR-GPS-SIM i am getting 3D position fix but i have a 3 things to discuss with you
[image]

atchyuth-rao, 2021-03-27:
[image]

Mictronics, 2021-03-27:
1) Unless you specify a specific station with --station, the -f command uses a random station while pulling the Rinex file. However, not all stations provide such file for the actual hour. Depends on time zone, local time etc. Stations are listed in gps.c.

2) & 3)
Getting a 3D fix with new coordinates takes more time when the receiver already knows the almanac and has seen a fix. Be patient and wait a significant time or cold start your GPS receiver.

atchyuth-rao, 2021-03-28:
1.I haven't changed any TCXO modification of my Pluto I think that is the one thing which I need to modify.

2. I want to repeat the same experiment with the dynamic coordinates(i.e.i want to simulate particular trajectory (path)).

For the above thing how should I Generate dynamic coordinates and how can I apply this coordinates to multi-sdr-gps-sim application

atchyuth-rao, 2021-03-30:
hi Micronics,

I am getting position fix with user motion file(user motion file is created with the help of satgen free software) but I have two things I need inputs from you

1.When we test with user motion the max duration is 300sec, can we increase this to max extend like for Static coordinates we can run up to 86400 sec, give me some inputs i will try.
2.I want test mutli-sdr-gps-sim on mobile phones,

I have generated GPS Signal with the help of Pluto SDR Tx connected with antenna and Trnamitted into free space,I turned ON my mobile phone location, I am getting fix with U-blox Receiver but my mobile cant changes its position.

Mictronics, 2021-03-30:
Simulation works for me with and dynamic on Samsung Galaxy S2 Tab and Motorola G6. However time to first fix depends on what position the device has seen before, relative to simulated coordinates.

atchyuth-rao, 2021-03-30:
Thanks man for your response

Forget about 3-D fix my mobile didn't get any Satellies(atleast PRN code--
blue colour bars), I have connected one antenna(came with PLUTO) to Pluto
SDR and Generated GPS signal, in my phone I have disabled sim card.

And one more thing can we simulate user motion more than 5 minutes, with
present alogorithm we can simulate max 5 min(for user motion)

atchyuth-rao, 2021-03-31:
> I don't know why you spam open PR's with your questions but to answer you
> last one: Yes the user motion can be extended, length is defined by
> USER_MOTION_SIZE.

I am getting output for user motion with more than 5 minutes.

1.i have only one pending work that is need to get mobile fix with Mobile
phones.

I will try and let you know.

atchyuth-rao, 2021-04-05:
Hi Mictronics,

have you ever tested Multi-sdr-gps-sim on drones using pluto SDR or Hackrf

atchyuth-rao, 2021-05-11:
> Your almanac is outdated. Older than 4 weeks?

I have tested with today brdc file from naasa website

Mictronics, 2021-05-11:
Can you provide me the file?

atchyuth-rao, 2021-05-12:
[brdc1320.21n.gz](https://github.com/osqzss/gps-sdr-sim/files/6463289/brdc1320.21n.gz)

Mictronics, 2021-05-12:
The file works. Your command line is wrong. You start the program with option ´-e rinex2.gz´ (the default) which is obviously outdated. Use ´-e brdc1320.21n.gz´ or any other file name in case of non-default Rinex files.

badrbouslikhin, 2021-12-28:
> The .grc file I posted is configured for a 4 MHz sample rate, and the xo_correction set to 40000000. That's all the help I can offer - I'm unsubscribing from this thread now.

Thanks for the grc file @jamesl-dm. It worked great for me. Although I am not able to sustain a fix for more than a few minutes (I can see both the smartphone and my nRF91 receiver) dropping fixes at the same time (usually around 4-5 mn). I tried to play with the ppm but I couldn't do better. Any tips?

deanZZZZZ, 2022-01-08:
Can any additional version be made for the OSX (mac)?
When compile the problem is with the processor thread ...