osqzss/gps-sdr-sim #229: Rapberry pi 3 .bin Corruption.

Commander-Crash, 2020-01-16:
Rapberry pi 3 B
OS Jessie
3A PSU
MicroSD is a Samsung evo 32GB U1
Kernel: 4.9.35-v7+ #1014 SMP Fri Jun 30 14:47:43 BST 2017 armv7l GNU/Linux

When Generating a GPS .bin file with the raspberry pi 3 B It will become corrupted somehow. GPS will show a 0.000 on lon/lat. Below is the command used.
./gps-sdr-sim -b 8 -e brdc0150.20n -l 51.5074,0.1278,100

Im not sure if this is a RPI issue SD-card issue or Kernel. I can generate just fine on a windows 7 PC and when I move the .bin file to the Rpi3 to be transmitted it works fine.

Any help on resolving this issue would be most appreciated.
Thanks

mikehibbett, 2020-01-23:
I see this too, on a Pi 2 model B.

The gpssim.bin file it truncated; 2GB rather than 3GB (approx)

The file contents start off the same as a file created on a PC, but blocks of data are missing.

Thanks for reporting the work-around, I can live without building the data file on the Pi.

whysoft, 2021-12-05:
I have a similar phenomenon here. 
This happens if I have too many processes running in the PI that require disk IO at the same time。The reason is the IO speed of SD card is limited