osqzss/gps-sdr-sim #230: Regarding the Intermediate Frequency (IF) and Sampling frequency values of generated GPSSIM.bin file

komalkumar6655, 2020-01-25:
Hello, 
I have generated the GPSSIM.bin file using GPS-SDR-SIM. Could you please tell me about  the Intermediate Frequency (IF) and Sampling frequency values of generated GPSSIM.bin file. So that it will be helpful. 
Expecting positive response from you.
Thank you in advance.

osqzss, 2020-01-25:
IF is zero Hz, and the sampling frequency is 2.6MHz by default.

komalkumar6655, 2020-01-25:
Thank you for your response .
But the problem is.
as putting the above mentioned values by you i cant get the acquisition results in GNSS sdr. Please help me with this.
[image]
[image]

osqzss, 2020-01-25:
I'm not familiar with GNSS-SDRLIB, but the data type should be int16.