osqzss/gps-sdr-sim #236: IEEE754 / rounded Navigation Message values

jonas229, 2020-02-17:
Hi,
i am trying to understand your code and have a question about the subframe generation. 
In your code you scale the values for the navigation message and afterwards cast them to long or unsigned long data types. 
Could you please explain where you got the information in which data type the values are transmitted inside the navigation message? 

I thought that all floating point values are transmitted without rounding. Therefore the values would be presented in the normal IEEE754 form, exponent and mantissa.

Am I missing something or isn't it necessary to transmit these values in a floating point value?

Greetings 
Jonas

osqzss, 2020-02-17:
Please refer to the GPS interface control documents. Everything you need to know is specified in the ICDs: https://www.gps.gov/technical/icwg/

jonas229, 2020-02-17:
I re-read the corresponding IS-GPS-200 File again. I think the sentence: "Parameters so indicated shall be two's complement, with the sign bit (+ or -) occupying the MSB" is helpful for me. The values are casted to long/signed long because of this? 
Is this correct or am i completely wrong ?

osqzss, 2020-02-18:
Every parameter in the navigation message has an integer value with a specific bit length and scale factor. See Table 20-III for example. These values has to be stuffed into  32-bit length word (long or signed long) frames specified in Figure 20-1.