osqzss/gps-sdr-sim #237: HackRF seems to send datas but smartphone receive nothing.

Akhoriz, 2020-02-20:
Hi ! 
I need your help with my GPS Spoofing for my Engineering Student Project.

I downloaded the last ephemeris on the Nasa's ftp.
Then created the .bin with: .\gps-sdr-sim.exe -b 8 -e .\brdc0510.20n -l -41.300988,174.775321,100
Then I sent the file to the HackRF with: .\hackrf_transfer.exe -t gpssim.bin -f 1575420000 -s 2600000 -a 1 -x 0

In the console I have this: 
call hackrf_set_sample_rate(2600000 Hz/2.600 MHz)
call hackrf_set_hw_sync_mode(0)
call hackrf_set_freq(1575420000 Hz/1575.420 MHz)
call hackrf_set_amp_enable(1)
Stop with Ctrl-C
 5.2 MiB / 1.002 sec =  5.2 MiB/second
 5.0 MiB / 1.002 sec =  5.0 MiB/second
 5.2 MiB / 1.003 sec =  5.2 MiB/second
 5.2 MiB / 1.003 sec =  5.2 MiB/second
 5.2 MiB / 1.003 sec =  5.2 MiB/second

My smartphone received nothing even if I remove the Wifi Scanning & Perfect Location

I also tried to scan the 1575,420 MHZ with GQRX and I had only noise.

Do you have an idea ?

kevtheskin, 2020-02-24:
Same here. Nothing :(

xxxxxaaaaaaaannnn, 2020-03-08:
Hello, I posted #240 that might help you.

MSK2032, 2022-03-26:
you, can try using a tiny TCXO (external oscillator). It worked for me to resolve no signal reception...atleast for GPS receivers. The issue of few GNSS receivers in android mobile receiving and still not accepting the spoofed signals, is yet to be resolved.