osqzss/gps-sdr-sim #239: Getting ephemeris from Ub-blox

ferorted, 2020-03-06:
I'm trying to obtain a Rinex broadcast ephemeris file to use instead of downloaded brdc.

I record RWAX messages from u-Center (connected to a U-blox MT8 receiver) for an hour.
Then I use RTKCONV to obtain the follow RINEX file.

[RAW-2.txt](https://github.com/osqzss/gps-sdr-sim/files/4299840/RAW-2.txt)

However It is only valid from 18:00 to 18:00h

[image]

How can I get a file valid for about 2 hour?

Mictronics, 2020-03-07:
I'm not familiar with RTKCONV, but from what I found in internet I guess you need to set start and end time manually.

ferorted, 2020-03-13:
Hi, I discovered that to obtain efemeris valid for 2h, i have to record more than 1h of data.

Imtiaz08, 2021-12-30:
@ferorted Can you please explain the process of how the spoofing signal is generated once we input our desired coordinates along with a rinex.nav file to the sdr sim? Or if you can provide me some reference material where I can study the whole process.

ferorted, 2022-01-14:
Hi,
I did my tests using pluto-sdr-sim project by Mictronics (based on gps-sdr-sim project). 
A Pluto SDR device is required, of course.
Now Mictronics has a new project which have some improvement and works very fine, and support HackRF, ADLAM-Pluto SDR devices: https://github.com/Mictronics/multi-sdr-gps-sim

Pay attention to Pluto SDR oscillator. you can find more information in https://www.mictronics.de/posts/ADLAM-Pluto-OCXO-Mod/