osqzss/gps-sdr-sim #243: Why "Carrier" IF Frequency

jonas229, 2020-04-05:
After reviewing the old tickets, i realized that the generated file generally contains baseband signals. Nevertheless the Nav+C/A data is multiplied with cosine/sine, and you mentioned that this the IF frequency centered at 0 Hz. I don't understand why this is necessary and why it depends on the Carrier Phase. 
Could you give me a source hint or a short explanation why the values are multiplied with the waveforms? 

Greetings
Jonas

osqzss, 2020-04-05:
The satellite motion creates a small frequency shift due to the Doppler effect.

jonas229, 2020-04-06:
Okay. So the Doppler frequency is taken into account at the code frequency when the next data Bit is computed and also at the carrier frequency? And the frequency shift of the Carrier is simulated with the multiplication with sine/cosine?

osqzss, 2020-04-06:
Yes.