osqzss/gps-sdr-sim #244: I can't figure out this script issue with my BladeRF

chownsauce, 2020-04-19:
[chown@mbp ~/Downloads/gps-sdr-sim (master *)]$ bladeRF-cli -l hostedxA9-latest.rbf
Loading fpga...
Successfully loaded FPGA bitstream!
[chown@mbp ~/Downloads/gps-sdr-sim (master *)]$ bladeRF-cli -s bladerf.script

For best results, it is not recommended to set both RX and TX to the
same frequency. Instead, consider offsetting them by at least 1 MHz
and mixing digitally.

For the above reason, 'set frequency <value>` is deprecated and
scheduled for removal in future bladeRF-cli versions.

Please use 'set frequency rx' and 'set frequency tx' to configure
channels individually.

  RX1 Frequency: 1575419998 Hz (Range: [70000000, 6000000000])
  RX2 Frequency: 1575419998 Hz (Range: [70000000, 6000000000])
  TX1 Frequency: 1575419998 Hz (Range: [47000000, 6000000000])
  TX2 Frequency: 1575419998 Hz (Range: [47000000, 6000000000])

  Setting RX1 sample rate - req:   2600000 0/1Hz, actual:   2600000 0/1Hz
  Setting RX2 sample rate - req:   2600000 0/1Hz, actual:   2600000 0/1Hz
  Setting TX1 sample rate - req:   2600000 0/1Hz, actual:   2600000 0/1Hz
  Setting TX2 sample rate - req:   2600000 0/1Hz, actual:   2600000 0/1Hz

  RX1 Bandwidth:   2500000 Hz (Range: [200000, 56000000])
  RX2 Bandwidth:   2500000 Hz (Range: [200000, 56000000])
  TX1 Bandwidth:   2500000 Hz (Range: [200000, 56000000])
  TX2 Bandwidth:   2500000 Hz (Range: [200000, 56000000])

  Error (bladerf.script:3): Invalid operation or parameter

osqzss, 2020-04-20:
https://www.nuand.com/frequently-asked-questions/

chownsauce, 2020-04-20:
What am I missing from here?

osqzss, 2020-04-21:
If you're using libbladeRF 2.0 or later, try "set gain tx ##" instead of "set txvga1 ##".
https://www.nuand.com/frequently-asked-questions/#How_do_I_set_gains
https://github.com/Nuand/bladeRF/wiki/bladeRF-CLI-Tips-and-Tricks#Receiving_samples

chownsauce, 2020-04-21:
Thank you! That helps a lot!

chownsauce, 2020-04-21:
This is what I'm getting:

[chown@mbp ~/Downloads/gps-sdr-sim-realtime (master *)]$ bladeRF-cli -i
bladeRF> ver

  bladeRF-cli version:        1.8.0-git-45521019
  libbladeRF version:         2.2.1-git-45521019

  Firmware version:           2.3.2
  FPGA version:               0.11.0 (configured by USB host)

bladeRF> quit
[chown@mbp ~/Downloads/gps-sdr-sim-realtime (master *)]$ bladeRF-cli -s bladerf.script

For best results, it is not recommended to set both RX and TX to the
same frequency. Instead, consider offsetting them by at least 1 MHz
and mixing digitally.

For the above reason, 'set frequency <value>` is deprecated and
scheduled for removal in future bladeRF-cli versions.

Please use 'set frequency rx' and 'set frequency tx' to configure
channels individually.

  RX1 Frequency: 1575419998 Hz (Range: [70000000, 6000000000])
  RX2 Frequency: 1575419998 Hz (Range: [70000000, 6000000000])
  TX1 Frequency: 1575419998 Hz (Range: [47000000, 6000000000])
  TX2 Frequency: 1575419998 Hz (Range: [47000000, 6000000000])

  Setting RX1 sample rate - req:   2600000 0/1Hz, actual:   2600000 0/1Hz
  Setting RX2 sample rate - req:   2600000 0/1Hz, actual:   2600000 0/1Hz
  Setting TX1 sample rate - req:   2600000 0/1Hz, actual:   2600000 0/1Hz
  Setting TX2 sample rate - req:   2600000 0/1Hz, actual:   2600000 0/1Hz

  RX1 Bandwidth:   2500000 Hz (Range: [200000, 56000000])
  RX2 Bandwidth:   2500000 Hz (Range: [200000, 56000000])
  TX1 Bandwidth:   2500000 Hz (Range: [200000, 56000000])
  TX2 Bandwidth:   2500000 Hz (Range: [200000, 56000000])

  Gain on RX1 cannot be changed while AGC is enabled.

  Error (bladerf.script:3): Operation invalid in current state

[chown@mbp ~/Downloads/gps-sdr-sim-realtime (master *)]$ cat bladerf.script
set frequency 1575.42M
set samplerate 2.6M
set bandwidth 2.5M
set gain rx 60
cal lms
cal dc tx
tx config file=gpssim.bin format=bin
tx start
tx wait

Hope that you can help.

chownsauce, 2020-04-21:
It's a bladeRF R9 is that helps you at all. I just don't know what I'm doing wrong.

chownsauce, 2020-04-21:
[chown@mbp ~/Downloads/gps-sdr-sim-realtime (master *)]$ bladeRF-cli -l ../hostedxA9-latest.rbf
Loading fpga...
Successfully loaded FPGA bitstream!

chownsauce, 2020-04-21:
My idea was that this script was supposed to work out of the box without any needed tweaks to the script but that does not seem to be the case. I've tried everything I know to try:

[chown@mbp ~/Downloads/gps-sdr-sim-realtime (master *)]$ cat bladerf.script
set frequency 1575.42M
set samplerate 2.6M
set bandwidth 2.5M
set agc off
set gain rx 60
cal lms
cal dc tx
tx config file=gpssim.bin format=bin
tx start
tx wait

osqzss, 2020-04-21:
Since I don't have a bladeRF micro 2.0, I can't be of much help. 
  
The error message mentioned about the AGC. Please try disable it with the "set agc off" command.
https://nuand.com/forums/viewtopic.php?t=8741

By the way, you need to set TX gain for the simulator, not RX.

chownsauce, 2020-04-21:
Here is what I've tried. I don't know what I'm missing. Sorry I'm a new to all of this.

[chown@mbp ~/Downloads/gps-sdr-sim-realtime (master *)]$ bladeRF-cli -l hostedxA9.rbf
Loading fpga...
Successfully loaded FPGA bitstream!
[chown@mbp ~/Downloads/gps-sdr-sim-realtime (master *)]$ bladeRF-cli -s bladerf.script

For best results, it is not recommended to set both RX and TX to the
same frequency. Instead, consider offsetting them by at least 1 MHz
and mixing digitally.

For the above reason, 'set frequency <value>` is deprecated and
scheduled for removal in future bladeRF-cli versions.

Please use 'set frequency rx' and 'set frequency tx' to configure
channels individually.

  RX1 Frequency: 1575419998 Hz (Range: [70000000, 6000000000])
  RX2 Frequency: 1575419998 Hz (Range: [70000000, 6000000000])
  TX1 Frequency: 1575419998 Hz (Range: [47000000, 6000000000])
  TX2 Frequency: 1575419998 Hz (Range: [47000000, 6000000000])

  Setting RX1 sample rate - req:   2600000 0/1Hz, actual:   2600000 0/1Hz
  Setting RX2 sample rate - req:   2600000 0/1Hz, actual:   2600000 0/1Hz
  Setting TX1 sample rate - req:   2600000 0/1Hz, actual:   2600000 0/1Hz
  Setting TX2 sample rate - req:   2600000 0/1Hz, actual:   2600000 0/1Hz

  RX1 Bandwidth:   2500000 Hz (Range: [200000, 56000000])
  RX2 Bandwidth:   2500000 Hz (Range: [200000, 56000000])
  TX1 Bandwidth:   2500000 Hz (Range: [200000, 56000000])
  TX2 Bandwidth:   2500000 Hz (Range: [200000, 56000000])

  RX1 AGC: Disabled
  RX2 AGC: Disabled

Setting TX1 overall gain to 60 dB
  Gain TX1 overall:   60 dB (Range: [-23.75, 66])
               dsa:  -90 dB (Range: [-89.75, 0])

  cal (bladerf.script:5): Only use this calibration on bladeRF x40 and x115 devices.

  If tuning mode is FPGA, consider setting tuning mode to host by running `set tuning_mode host`.

  Error (bladerf.script:6): Operation not supported

[chown@mbp ~/Downloads/gps-sdr-sim-realtime (master *)]$ cat bladerf.script
set frequency 1575.42M
set samplerate 2.6M
set bandwidth 2.5M
set agc off
set gain tx1 60
cal lms
cal dc tx
tx config file=gpssim.bin format=bin
tx start
tx wait

osqzss, 2020-04-21:
You can skip "cal lms" and "cal dc tx" lines and see how things go.

chownsauce, 2020-04-21:
Thanks!! That seemed to do the trick.

[chown@mbp ~/Downloads/gps-sdr-sim-realtime (master *)]$ bladeRF-cli -s bladerf.script

For best results, it is not recommended to set both RX and TX to the
same frequency. Instead, consider offsetting them by at least 1 MHz
and mixing digitally.

For the above reason, 'set frequency <value>` is deprecated and
scheduled for removal in future bladeRF-cli versions.

Please use 'set frequency rx' and 'set frequency tx' to configure
channels individually.

  RX1 Frequency: 1575419998 Hz (Range: [70000000, 6000000000])
  RX2 Frequency: 1575419998 Hz (Range: [70000000, 6000000000])
  TX1 Frequency: 1575419998 Hz (Range: [47000000, 6000000000])
  TX2 Frequency: 1575419998 Hz (Range: [47000000, 6000000000])

  Setting RX1 sample rate - req:   2600000 0/1Hz, actual:   2600000 0/1Hz
  Setting RX2 sample rate - req:   2600000 0/1Hz, actual:   2600000 0/1Hz
  Setting TX1 sample rate - req:   2600000 0/1Hz, actual:   2600000 0/1Hz
  Setting TX2 sample rate - req:   2600000 0/1Hz, actual:   2600000 0/1Hz

  RX1 Bandwidth:   2500000 Hz (Range: [200000, 56000000])
  RX2 Bandwidth:   2500000 Hz (Range: [200000, 56000000])
  TX1 Bandwidth:   2500000 Hz (Range: [200000, 56000000])
  TX2 Bandwidth:   2500000 Hz (Range: [200000, 56000000])

  RX1 AGC: Disabled
  RX2 AGC: Disabled

Setting TX1 overall gain to 60 dB
  Gain TX1 overall:   60 dB (Range: [-23.75, 66])
               dsa:  -90 dB (Range: [-89.75, 0])

^C[chown@mbp ~/Downloads/gps-sdr-sim-realtime (master *)]$ ll

chownsauce, 2020-04-21:
Is there a reason why Snapchat doesn't show the new location? Maybe their algorithm has changed a bit? I haven't done this since hackrf days and it used to work fine. Just wondering what you thought.

Blystad, 2020-05-08:
@chownsauce Can you confirm that you got it to work with a bladeRF 2.0 micro? Is it the A4 or A9 you have? (and does it matter)

With regards to Snapchat, you might fall victim to the Network Location Provider providing a location, and your GPS simulator providing another location. Try to disable the Google Network Location Provider (assuming you're running on Android, on iOS I don't know if it's possible to disable the Apple network location provider). NLP uses WiFi networks and cell towers to provide an approximate location.

czbxzm, 2023-09-05:
hello my friend,have fixed this question caused by bladerf xa4(2.0) ? May i get ur script?