osqzss/gps-sdr-sim #245: No GPS fix whatever I do

majedalanni, 2020-05-03:
Hi,
I have HackRF one with ANT500, trying to spoof my android 10 phone GPS, I follow exactly instruction and couple videos, removed my sim card turned high accuracy off in Android, turn on airplane mode. but it will never fix 

my android is rooted, is there a way we can check logs maybe to see why it not fix or do I do something wrong?

Thanks

[image]
[image]
[image]
[image]

michalsternadel, 2020-05-04:
Try to generate fake signal with **-b 8** option like this:
`gps-sdr-sim -b 8 -e brdc1090.20n -l 55.2768359,6.2364598,100`
then push it to the air using:
`hackrf_transfer -t gpssim.bin -f 1575420000 -s 2600000 -a 1 -x 0`
As far I know HackRF One requires an 8 bit signed IQ raw file with no header information.
Works for me.

majedalanni, 2020-05-04:
> Try to generate fake signal with **-b 8** option like this:
> `gps-sdr-sim -b 8 -e brdc1090.20n -l 55.2768359,6.2364598,100`
> then push it to the air using:
> `hackrf_transfer -t gpssim.bin -f 1575420000 [...]

Thanks you sir, I already did -b 8 with no success, however I started to get one device locked after I add -C -10

RannyTheCoder, 2020-07-23:
@majedalanni ..Is your lock stable for the duration of simulation (default 300 secs) ?

BorisDess, 2020-10-26:
Same, I tried with two different 

I'm inside without any GPS signal.
When launching signal transmission.
In view I see tons of GPS (30+).
Same setup that the post : mi9 android phone, airplane mode and every A-gps feature disabled.
I'm also using an IoT device with M8N module + Taoglass antenna ; no fix either.

Tried -C -10 without result.
I'm using a Chinese clone of the Hackrf with a dumb wire antenna.
Phone and IoT device are on the antenna.

Imtiaz08, 2022-05-20:
what does the -x represents please?